hypriot/image-builder-rpi #4: Is heredoc correct?

StefanScherer, 2016-01-10:
I wonder if this heredoc works in line 61 https://github.com/hypriot/image-builder-rpi/blob/master/build.sh#L61

```
chroot ${BUILD_PATH} /bin/bash <<"EOCHROOT"
```

as it should expand variables for FSTAB in line 84 https://github.com/hypriot/image-builder-rpi/blob/master/build.sh#L84

```
  echo ${FSTAB} > /etc/fstab
```

I thought expansion needs a heredoc without the quotes.
I'll test inside a vagrant box.

DieterReuter, 2016-01-10:
The heredoc has to be this way with chroot! So, that's absolute correct in  https://github.com/hypriot/image-builder-rpi/blob/master/build.sh#L61. Without the quotes in `<<"EOCHROOT"` you'll get an error.

StefanScherer, 2016-01-10:
So we can pin all the kernel packages by using the KERNEL_* variables, that part is missing ;-)

StefanScherer, 2016-01-10:
I have added a test to proof that /etc/fstab is correct in #3

StefanScherer, 2016-01-10:
@DieterReuter Are you sure? The test in build https://travis-ci.org/hypriot/image-builder-rpi/builds/101444491 says "empty" /etc/fstab.

StefanScherer, 2016-01-18:
resolved in the meantime

closed 2016-01-18 by StefanScherer