hypriot/image-builder-rpi #18: docker-machine-hypriot 0.4.1 findings

StefanScherer, 2016-01-31:
I've tried to use `docker-machine-hypriot` v0.4.1 to connect to the new HypriotOS v0.1.1 to check if it's compatible. After some several errors and reflashing to have a first boot experience again and again I found a way with additional options.

## Reproduce
What I did: 

1. Add my SSH pub key to the new machine,
2. then run `docker-machine` with
3. the IP of the RPi
4. and the username `pirate` as the root login is disabled
5. and the `overlay` storage driver as the default `aufs` isn't available on RPi

```
$ ssh-copy-id pirate@192.168.10.23
$ docker-machine-hypriot create -d hypriot --hypriot-ip-address 192.168.10.23 --hypriot-ssh-user pirate --engine-storage-driver overlay hypriotos
```

## Findings
Some interesting aspect can be found in the debug output (see below for all details):

* the additional keys in `/etc/os-release` result in a warning:
```
Couldn't set key HYPRIOT_OS, no corresponding struct field found
Couldn't set key HYPRIOT_TAG, no corresponding struct field found
```
* The RPi is detected as `debian` and not `raspbian`. So our hypriot provisioner in that static `docker-machine-hypriot` binary isn't used, but the standard (Intel-only tested) debian provisioner.
```
found compatible host: debian
```
* As a result it would use the following script if `docker` wasn't installed already:
```
if ! type docker; then curl -sSL https://get.docker.com | sh -; fi
```
* It writes a new config to `/etc/systemd/system/docker.service` but it still uses the `-d` option. So this old docker-machine binary won't work with Docker 1.10.
* The docker daemon does not start as the standard storage driver uses `aufs` and does not fit our RPi drivers. S
```
$ sudo tail -50 /var/log/syslog
Jan 31 08:25:18 black-pearl systemd[1]: Starting docker.service...
Jan 31 08:25:18 black-pearl systemd[1]: Failed to reset devices.list on /system.slice: Invalid argument
Jan 31 08:25:18 black-pearl systemd[1]: Started docker.service.
Jan 31 08:25:18 black-pearl docker[1725]: Warning: '-d' is deprecated, it will be removed soon. See usage.
Jan 31 08:25:18 black-pearl docker[1725]: time="2016-01-31T08:25:18Z" level=warning msg="please use 'docker daemon' instead."
Jan 31 08:25:24 black-pearl docker[1725]: time="2016-01-31T08:25:24.416688572Z" level=fatal msg="Error starting daemon: error initializing graphdriver: driver not supported"
Jan 31 08:25:24 black-pearl systemd[1]: docker.service: main process exited, code=exited, status=1/FAILURE
Jan 31 08:25:24 black-pearl systemd[1]: Unit docker.service entered failed state.
```

## Debug output
Here is the complete debug output:

```
$ ssh-copy-id pirate@black-pearl.local
/usr/local/bin/ssh-copy-id: INFO: attempting to log in with the new key(s), to filter out any that are already installed
/usr/local/bin/ssh-copy-id: INFO: 1 key(s) remain to be installed -- if you are prompted now it is to install the new keys
pirate@black-pearl.local's password: 

Number of key(s) added:        1

Now try logging into the machine, with:   "ssh 'pirate@black-pearl.local'"
and check to make sure that only the key(s) you wanted were added.

~
$ docker-machine-hypriot -D create -d hypriot --hypriot-ip-address 192.168.10.23 --hypriot-ssh-user pirate --engine-storage-driver overlay hypriotos
Importing SSH key...
IP: 192.168.10.23
Getting to WaitForSSH function...
Testing TCP connection to: 192.168.10.23:22
Using SSH client type: external
About to run SSH command:
exit 0
&{/usr/bin/ssh [/usr/bin/ssh -o PasswordAuthentication=no -o IdentitiesOnly=yes -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -o LogLevel=quiet -o ConnectionAttempts=3 -o ConnectTimeout=10 -i /Users/stefan/.docker/machine/machines/hypriotos/id_rsa -p 22 pirate@192.168.10.23 exit 0] []  <nil> <nil> <nil> [] <nil> <nil> <nil> <nil> false [] [] [] [] <nil>}
[... 242 lines cut ...]
Daemon not responding yet: dial tcp 192.168.10.23:2376: connection refused
configuring swarm
enabling docker in systemd
Using SSH client type: external
About to run SSH command:
sudo systemctl daemon-reload
&{/usr/bin/ssh [/usr/bin/ssh -o PasswordAuthentication=no -o IdentitiesOnly=yes -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -o LogLevel=quiet -o ConnectionAttempts=3 -o ConnectTimeout=10 -i /Users/stefan/.docker/machine/machines/hypriotos/id_rsa -p 22 pirate@192.168.10.23 sudo systemctl daemon-reload] []  <nil> <nil> <nil> [] <nil> <nil> <nil> <nil> false [] [] [] [] <nil>}
SSH cmd err, output: <nil>: 
Using SSH client type: external
About to run SSH command:
sudo systemctl enable docker
&{/usr/bin/ssh [/usr/bin/ssh -o PasswordAuthentication=no -o IdentitiesOnly=yes -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -o LogLevel=quiet -o ConnectionAttempts=3 -o ConnectTimeout=10 -i /Users/stefan/.docker/machine/machines/hypriotos/id_rsa -p 22 pirate@192.168.10.23 sudo systemctl enable docker] []  <nil> <nil> <nil> [] <nil> <nil> <nil> <nil> false [] [] [] [] <nil>}
SSH cmd err, output: exit status 1: Synchronizing state for docker.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d docker defaults
Executing /usr/sbin/update-rc.d docker enable
Failed to execute operation: File exists

SSH cmd error!
command: sudo systemctl enable docker
err    : exit status 1
output : Synchronizing state for docker.service with sysvinit using update-rc.d...
Executing /usr/sbin/update-rc.d docker defaults
Executing /usr/sbin/update-rc.d docker enable
Failed to execute operation: File exists
```

StefanScherer, 2016-01-31:
Using the latest `docker-machine` v0.6.0-rc2 binary also works "out-of-the-box" with some additional options:

```bash
docker-machine -D create -d generic --generic-ip-address 192.168.10.23 --generic-ssh-user pirate --engine-storage-driver overlay hypriotos
```

Creating a swarm probably needs the `--swarm-image hypriot/rpi-swarm:latest`, but I haven't tried it yet.

Ping @benediktarnold :-)

StefanScherer, 2016-01-31:
Oh, there's another interesting option: `--engine-install-url "https://get.docker.com"`

So we could have a workaround with something like `--engine-install-url "https://get.hypriot.com"` :smile:

jottr, 2016-04-03:
Hopefully I'm not spamming:
If I read @StefanScherers post correctly, using the official `docker-machine` binary from `0.6.0-rc2` should be working out of the box. 
I've tried with `0.6.0, build e27fb87` and `0.7.0-rc1`:

```bash
HypriotOS: root@jupiter in ~
$ docker --version
Docker version 1.10.3, build 20f81dd
```

```bash
/tmp/docker-machine-Darwin-x86_64 -D create -d generic --generic-ip-address 192.168.2.134  --generic-ssh-key /Users/jottr/.ssh/id_rsa --engine-storage-driver overlay rpi

Docker Machine Version:  0.7.0-rc1, build 0fb68ca
Found binary path at /tmp/docker-machine-Darwin-x86_64
Launching plugin server for driver generic
Plugin server listening at address 127.0.0.1:54347
() Calling .GetVersion
Using API Version  1
() Calling .SetConfigRaw
() Calling .GetMachineName
(flag-lookup) Calling .GetMachineName
(flag-lookup) Calling .DriverName
(flag-lookup) Calling .GetCreateFlags
Found binary path at /tmp/docker-machine-Darwin-x86_64
Launching plugin server for driver generic
Plugin server listening at address 127.0.0.1:54351
() Calling .GetVersion
Using API Version  1
() Calling .SetConfigRaw
() Calling .GetMachineName
(rpi) Calling .GetMachineName
(rpi) Calling .DriverName
(rpi) Calling .GetCreateFlags
(rpi) Calling .SetConfigFromFlags
Running pre-create checks...
(rpi) Calling .PreCreateCheck
(rpi) Calling .GetConfigRaw
Creating machine...
(rpi) Calling .Create
(rpi) Importing SSH key...
(rpi) Calling .GetConfigRaw
(rpi) DBG | IP: 192.168.2.134
(rpi) Calling .DriverName
(rpi) Calling .DriverName
Waiting for machine to be running, this may take a few minutes...
(rpi) Calling .GetState
Detecting operating system of created instance...
Waiting for SSH to be available...
Getting to WaitForSSH function...
(rpi) Calling .GetSSHHostname
(rpi) Calling .GetSSHPort
(rpi) Calling .GetSSHKeyPath
(rpi) Calling .GetSSHKeyPath
(rpi) Calling .GetSSHUsername
Using SSH client type: external
&{[-o BatchMode=yes -o PasswordAuthentication=no -o StrictHostKeyChecking=no -o UserKnownHostsFile=/dev/null -o LogLevel=quiet -o ConnectionAttempts=3 -o ConnectTimeout=10 -o ControlMaster=no -o ControlPath=none root@192.168.2.134 -o IdentitiesOnly=yes -i /Users/jottr/.docker/machine/machines/rpi/id_rsa -p 22] /usr/bin/ssh <nil>}
About to run SSH command:
exit 0
SSH cmd err, output: exit status 255:
Error getting ssh command 'exit 0' : Something went wrong running an SSH command!
command : exit 0
err     : exit status 255
output  :
```

StefanScherer, 2016-04-04:
@jottr The new SD image does not have a SSH login for root, only user pirate. You have to specify `--generic-ssh-user pirate` as well.

jottr, 2016-04-04:
@StefanScherer so it seems I **did** indeed hijack this issue. I'm using the `0.7.0 Berry (beta)` avalailable on the download page. Sorry for that. 

The `pirate` user is not available. After copying my ssh key, I am able to ssh on to the pi as a root user. 
Trying to to the same thing with 
```
docker-machine -D create -d generic \
--generic-ip-address 192.168.2.134 \
--generic-ssh-user root \
--engine-storage-driver overlay rpi
```
results in the following error:
`Error creating machine: Error detecting OS: OS type not recognized`

StefanScherer, 2016-04-04:
@jottr No problem :smiley: 

OK, for the 0.6/0.7 HypriotOS images based on Raspbian you have to prepare the OS a little bit to work with standard docker-machine binary. Have a look at the blog post http://blog.hypriot.com/post/how-to-setup-rpi-docker-swarm/ beginning with **Prepare your Pi’s for docker-machine**.

In /etc/os-release the string raspbian must be changed to debian.

closed 2016-04-25 by StefanScherer