hypriot/image-builder-rpi #31: Switching SD card between RPi's: eth1: renamed from eth0

StefanScherer, 2016-04-08:
When I boot a RPi with the v0.4.0 SD card image, shutdown and plug the SD card into another RPi, then the network does not work.

In /var/log/messages I see a message like

```
eth1: renamed from eth0
```

Probably due to the different MAC addresses of the RPi's

DieterReuter, 2016-04-08:
Just tested it, booting RPi3 first and then used the same SD card in a RPi2. Here are the messages from `dmesg` command:
```
$ dmesg | grep eth
[    3.139465] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-3f980000.usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:d5:98:5b
[    5.653463] smsc95xx 1-1.1:1.0 eth1: renamed from eth0
[    5.745601] systemd-udevd[112]: renamed network interface eth0 to eth1
```

DieterReuter, 2016-04-08:
This could be easily fixed in `/etc/systemd/network/eth0.network`, just use 
```
[Match]
Name=eth*
```
instead of `Name=eth1`.

DieterReuter, 2016-04-08:
Can be fixed with https://github.com/hypriot/os-rootfs/pull/25.

PR #40 (2016-04-09): Upgrade versions, install WiFi firmware and further improvements
* Upgrade HYPRIOT_OS_VERSION to v0.8.4 (fixes #31 and #33)
* Upgrade docker-hypriot to 1.10.3-1 (fixes #37)
* Upgrade docker-compose to 1.6.2-28 (fixes #36)
* Install WiFi firmware packages (fixes #21 and #35)
* Add a default config.txt file (fixes #34)
* Add tests for Raspberry Pi boot files
closed 2016-04-09 by DieterReuter