hypriot/image-builder-rpi #35: WiFi firmware and related integration tests are missing.

DieterReuter, 2016-04-08:
In the old integration tests we had some checks if the WiFi firmware packages are installed. These tests are missing and the WiFi modules are not installed in v0.4.0. This issue is related to #21.

DieterReuter, 2016-04-08:
Here is a complete list of available firmware packages:
```
apt-cache search firmware- | grep "^firmware-"
firmware-linux-free - Binary firmware for various drivers in the Linux kernel
firmware-b43-installer - firmware installer for the b43 driver
firmware-b43legacy-installer - firmware installer for the b43legacy driver
firmware-adi - Binary firmware for Analog Devices Inc. DSL modem chips
firmware-atheros - Binary firmware for Atheros wireless cards
firmware-bnx2 - Binary firmware for Broadcom NetXtremeII
firmware-bnx2x - Binary firmware for Broadcom NetXtreme II 10Gb
firmware-brcm80211 - Binary firmware for Broadcom 802.11 wireless cards
firmware-crystalhd - Crystal HD Video Decoder (firmware)
firmware-intelwimax - Binary firmware for Intel WiMAX Connection
firmware-ipw2x00 - Binary firmware for Intel Pro Wireless 2100, 2200 and 2915
firmware-ivtv - Binary firmware for iTVC15-family MPEG codecs (ivtv and pvrusb2 drivers)
firmware-iwlwifi - Binary firmware for Intel Wireless cards
firmware-libertas - Binary firmware for Marvell Libertas 8xxx wireless cards
firmware-linux - Binary firmware for various drivers in the Linux kernel (meta-package)
firmware-linux-nonfree - Binary firmware for various drivers in the Linux kernel
firmware-myricom - Binary firmware for Myri-10G Ethernet adapters
firmware-netxen - Binary firmware for QLogic Intelligent Ethernet (3000 and 3100 Series)
firmware-qlogic - Binary firmware for QLogic IBA7220, QLA1xxx, ISP2xxx and SP2x2
firmware-ralink - Binary firmware for Ralink wireless cards
firmware-realtek - Binary firmware for Realtek wired and wireless network adapters
firmware-samsung - Binary firmware for Samsung MFC video codecs
firmware-ti-connectivity - Binary firmware for TI Connectivity wireless network adapters
firmware-zd1211 - binary firmware for the zd1211rw wireless driver
```
I guess not all of these are WiFi related, but we can easily select the relevant packages for now.

DieterReuter, 2016-04-08:
To list all the installed firmware packages we could use the command:
```
dpkg -l | grep "^ii  firmware-"
ii  firmware-ralink                    0.43                      all          Binary firmware for Ralink wireless cards
ii  firmware-realtek                   0.43                      all          Binary firmware for Realtek wired and wireless network adapters
```
We should run this on a Raspbian LITE image to determine all the installed packages we should included into HypriotOS, too.

DieterReuter, 2016-04-08:
Ok, here is the firmware list which is already installed on a Raspbian LITE `2016-03-18-raspbian-jessie-lite.img`:
```
dpkg -l | grep "^ii  firmware-"
ii  firmware-atheros               0.43+rpi4                  all          Binary firmware for Atheros wireless cards
ii  firmware-brcm80211             0.43+rpi4                  all          Binary firmware for Broadcom 802.11 wireless cards
ii  firmware-libertas              0.43+rpi4                  all          Binary firmware for Marvell Libertas 8xxx wireless cards
ii  firmware-ralink                0.43+rpi4                  all          Binary firmware for Ralink wireless cards
ii  firmware-realtek               0.43+rpi4                  all          Binary firmware for Realtek wired and wireless network adapters
```

Govinda-Fichtner, 2016-04-09:
@firecyberice and I am working on getting the WiFi working...

DieterReuter, 2016-04-09:
This issues is already fixed by #40.

closed 2016-04-09 by DieterReuter