hypriot/image-builder-rpi #39: Updating package.io lists results in errors

DieterReuter, 2016-04-08:
Updating the Schatzkiste package lists leads to an error.
```
curl -s https://packagecloud.io/install/repositories/Hypriot/Schatzkiste/script.deb.sh | sudo bash
curl -s https://packagecloud.io/install/repositories/Hypriot/Schatzkiste/script.deb.sh | sudo bash
Detected operating system as hypriotos/8.
Checking for curl...
Detected curl...
Running apt-get update... done.
Installing apt-transport-https... done.
Installing /etc/apt/sources.list.d/Hypriot_Schatzkiste.list...curl: (22) The requested URL returned error: 404 Not Found

Unable to download repo config from: https://packagecloud.io/install/repositories/Hypriot/Schatzkiste/config_file.list?os=hypriotos&dist=8&source=script

This usually happens if your operating system is not supported by
packagecloud.io, or this script's OS detection failed.

You can override the OS detection by setting os= and dist= prior to running this script.
You can find a list of supported OSes and distributions on our website: https://packagecloud.io/docs#os_distro_version

For example, to force Ubuntu Trusty: os=ubuntu dist=trusty ./script.sh

If you are running a supported OS, please email support@packagecloud.io and report this.
```

DieterReuter, 2016-04-09:
We should include a `/etc/lsb-release` file, then the Package.io script can detect `raspbian/jessie`. Should be tested on the other ARM boards as well.

StefanScherer, 2016-08-29:
fixed with https://github.com/hypriot/image-builder-rpi/commit/53bee9e63f905d1a92d8056cfc64b47b4b11da7d

closed 2016-08-29 by StefanScherer