hypriot/image-builder-rpi #72: Bluetooth doesn't work on Pi3

DieterReuter, 2016-05-14:
With SD image v0.5.9 the BT doesn't work. On the serial UART console I'm getting an error
```
[FAILED] Failed to start Configure Bluetooth Modems connected by UART.
See 'systemctl status hciuart.service' for details.
```

Issuing `systemctl status hciuart.service` reveals some problems on starting the service:
```
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/lib/systemd/system/hciuart.service; enabled)
   Active: failed (Result: exit-code) since Sat 2016-05-14 15:09:49 UTC; 12s ago
  Process: 8357 ExecStart=/usr/bin/hciattach /dev/serial1 bcm43xx 921600 noflow - (code=exited, status=1/FAILURE)
 Main PID: 4457 (code=exited, status=0/SUCCESS)

May 14 15:09:49 black-pearl hciattach[8357]: Can't open serial port: No such file or directory
May 14 15:09:49 black-pearl hciattach[8357]: Can't initialize device: No such file or directory
May 14 15:09:49 black-pearl systemd[1]: hciuart.service: control process exited, code=exited status=1
May 14 15:09:49 black-pearl systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
May 14 15:09:49 black-pearl systemd[1]: Unit hciuart.service entered failed state.
```

The device `/dev/serial1` doesn't exist, so the hciuart.service failed to start.

**Possible solution:** using device `/dev/ttyAMA0` seems to fix the problem. But we have to investigate if this is the best way, before using the fix.

Govinda-Fichtner, 2016-05-14:
@DieterReuter It does work for me when I run the integration tests. No failing bluetooth tests.

DieterReuter, 2016-05-16:
There is a change introduced in package "pi-bluetooth=0.1.1" which is incompatible with HypriotOS, this causes a failure on starting up the Bluetooth service. Package "pi-bluetooth=0.1.0" is not available any more, so all new builds will end up having an issue with BT.

I'm now testing a work around to fix this issue, so we can provide a working Bluetooth device until we get a better solution.

PR #78 (2016-05-16): Fix bluetooth service start on Pi3
Fixes #72.
commit 36d125bf (2016-05-16): Fix bluethooth service start on Pi3

Fixes #72.
closed 2016-05-16 by DieterReuter