hypriot/image-builder-rpi #80: Build 0.5.14 corrupted if docker-machine create -d generic run on it

richtong, 2016-05-17:
If you create a 0.5.14 (I also tried on 0.5.5) SD, then if you try to change /etc/os-release to debian and then run

```
docker-machine create -d generic --generic-ip-address <insert here> --generic-ssh-user pirate
```

Then you get a corrupted docker installation. Running `docker ps` locally gives you:

```
Cannot connect to the Docker daemon. Is the docker daemon running on this host?
```

If you try `sudo systemctl restart docker` then you get:

```
An error occurred trying to connect: Get http:///var/run/docker.sock/v1.23/containers/json: read unix /var/run/docker.sock: connection reset by peer
```

And if you try `sudo apt-get install hypriot-docker` and you have your repos installed, you get it trying to install the older 1.10.3 driver:

```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following NEW packages will be installed:
  docker-hypriot
0 upgraded, 1 newly installed, 0 to remove and 5 not upgraded.
Need to get 6,608 kB of archives.
After this operation, 32.3 MB of additional disk space will be used.
Get:1 https://packagecloud.io/Hypriot/Schatzkiste/debian/ wheezy/main docker-hypriot armhf 1.10.3-1 [6,608 kB]
Fetched 6,608 kB in 5s (1,143 kB/s)           
Selecting previously unselected package docker-hypriot.
(Reading database ... 48754 files and directories currently installed.)
Preparing to unpack .../docker-hypriot_1.10.3-1_armhf.deb ...
Unpacking docker-hypriot (1.10.3-1) ...
dpkg: error processing archive /var/cache/apt/archives/docker-hypriot_1.10.3-1_armhf.deb (--unpack):
 trying to overwrite '/lib/systemd/system/docker.socket', which is also in package docker-engine 1.11.1-1
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/docker-hypriot_1.10.3-1_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

I'm a little stuck as I'm not a deep docker-machine person, but seems like the collision is between the 1.11 that is installed already and docker-machine trying to install something else.

StefanScherer, 2016-05-17:
Just for the records: With 0.5.4 we introduced Docker 1.11.1,
a VirtualBox machine 1.11.1 also does not contain any daemon.json file.

richtong, 2016-05-17:
Oops fixed the version numbers should say 0.5.14. I know 0.5.2 works for sure.

richtong, 2016-05-17:
Also wants sure if I should post to this or to docker-hypriot, but `apt-get install docker-hypriot` is also broken. Still on 1.10 and it fails to install properly with 0.5.14. I haven't had time to investigate, but perhaps you should bump release that.

StefanScherer, 2016-05-17:
@richtong To install docker-hypriot I guess you have to remove the package docker-engine manually. There are no dependencies between these two packages.

StefanScherer, 2016-05-18:
The /etc/docker/daemon.json was introduced in v0.5.5 with commit https://github.com/hypriot/image-builder-rpi/commit/f09778ce7deda95fceb8b712d04b597f73191e0a

We have to remove it for now to make the default SD image docker-machine compatible. Overlay and local unix socket can be achieved w/o this file as well.

StefanScherer, 2016-05-18:
Instead of removing /et/docker/daemon.json we can just make it empty with only 

```
{
}
```

This solves the docker-machine problem.

closed 2016-05-18 by StefanScherer

richtong, 2016-05-18:
Thanks I see sounds like the blog needs to deprecate docker-hypriot and use docker-engine instead then. That is way more standard! Nice work testing 0.5.15 now.

PR #82 (2016-05-18): Use empty daemon.json to make it compatible w/ docker-machine
Is an empty /etc/docker/daemon.json and Docker engine still uses its defaults with overlay and local unix socket.

This makes docker-machine work with our SD image again until docker-machine is capable parsing and updating this daemon.json (docker/machine#3062)

Fixes #80