hypriot/image-builder-rpi #85: Docker Engine uses devicemapper instead of overlay

StefanScherer, 2016-05-18:
With v0.5.15 the docker engine uses devicemapper instead of overlay.

Even with an entry `overlay` in `/etc/modules` it still defaults to devicemapper.
So we have to add this into the systemd config file, but NOT daemon.json.

StefanScherer, 2016-05-18:
That's what I find after a `docker machine create ...` on a v0.5.15:

```
$ find . -print  | grep dock
./system/multi-user.target.wants/docker.service
./system/docker.service
./system/sockets.target.wants/docker.socket
HypriotOS/armv7: pirate@barbossa in /etc/systemd
$ cat system/multi-user.target.wants/docker.service 
[Service]
ExecStart=/usr/bin/docker daemon -H tcp://0.0.0.0:2376 -H unix:///var/run/docker.sock --storage-driver overlay --tlsverify --tlscacert /etc/docker/ca.pem --tlscert /etc/docker/server.pem --tlskey /etc/docker/server-key.pem --label provider=generic 
MountFlags=slave
LimitNOFILE=1048576
LimitNPROC=1048576
LimitCORE=infinity
Environment=

[Install]
WantedBy=multi-user.target
HypriotOS/armv7: pirate@barbossa in /etc/systemd
$ cat system/docker.service 
[Service]
ExecStart=/usr/bin/docker daemon -H tcp://0.0.0.0:2376 -H unix:///var/run/docker.sock --storage-driver overlay --tlsverify --tlscacert /etc/docker/ca.pem --tlscert /etc/docker/server.pem --tlskey /etc/docker/server-key.pem --label provider=generic 
MountFlags=slave
LimitNOFILE=1048576
LimitNPROC=1048576
LimitCORE=infinity
Environment=

[Install]
WantedBy=multi-user.target
```

StefanScherer, 2016-05-18:
ping @Govinda-Fichtner /etc/systemd/system/docker.service

Govinda-Fichtner, 2016-05-18:
@StefanScherer in which directory did you issue the find command?

StefanScherer, 2016-05-18:
In /etc/systemd

BTW we should improve the bash prompt 😜

StefanScherer, 2016-05-18:
Sorry, our bash prompt shows the current dir, see above.

StefanScherer, 2016-05-19:
resolved with #86, release v0.5.16

closed 2016-05-19 by StefanScherer