hypriot/image-builder-rpi #105: Prepare next HypriotOS release "Blackbeard"

DieterReuter, 2016-07-23:
This is the task list for the next HypriotOS release "Blackbeard" for all Raspberry Pi models. Docker is preparing a major release with Docker Engine 1.12 and is also updating all the other tools, so we'll update HypriotOS to use all the new Docker tools as soon as possible. All changes should be done in single pull requests in order to keep them small and independently.

- [ ] update Linux kernel 4.4.15, https://github.com/hypriot/rpi-kernel/pull/25
- [ ] update docker-engine 1.12.0
- [ ] update docker-machine 0.8.0
- [ ] update docker-compose
- [ ] update rootfs
  - [ ] switch back to `ID=raspbian` in `/etc/os-release` (if docker-machine supports raspbian)
- [ ] update device-init
- [ ] remove hypriot-cluster-lab
- [ ] remove pre-installed consul image
- [ ] remove pre-installed swarm image

Govinda-Fichtner, 2016-07-23:
@DieterReuter I wonder if it still makes sense to leave the Cluster Lab as part of the image as the Swarm Mode and Cluster Lab should work side by side. Cluster Lab uses the old but not deprecated way of using Swarm with Consul... 

If people still want to use it, it is still there. If not it is not in the way of using Docker Swarm mode.

What do you think?

DieterReuter, 2016-07-23:
@Govinda-Fichtner We could leave it for now, and decide later whether we want to keep it or drop it. I'll change it to an optional task.

DieterReuter, 2016-08-19:
Great to close long running issue. 😄

closed 2016-08-19 by DieterReuter