hypriot/image-builder-rpi #107: Add /sbin to priate PATH

marcosnils, 2016-07-28:
By default many non-root specific utilities are there. It'd be nice if the base image already includes this path in the PATH variable.

DieterReuter, 2016-07-28:
@marcosnils what about just using `sudo` to use those commands! I'm assuming this should be the preferred way. But we can discuss this further.

marcosnils, 2016-07-28:
@DieterReuter there shouldn't be a need to use sudo for commands that can be executed by the user?. 

From a usability perspective is also better, I thought that some tools weren't installed as I kept getting `command not found` but the binaries were there, just not in the path, so I had to do a `find /` to realize about it

I've discussed this with @StefanScherer in Slack and he told me to file the issue.

StefanScherer, 2016-07-28:
Yes to start the discussion and have feedback from real users :-)
I don't remember how eg. standard Raspbian behaves.

DieterReuter, 2016-07-28:
Let's just figure out the default behaviour of Raspbian and making some detailed tests, then we do have some more background to decide if we could easily improve the user experience.

marcosnils, 2016-07-28:
@DieterReuter this is my first rpi experience and I just started an hour ago. I'm not familiarized with raspbian or it's directory structure but when I used it before installing hypriot, I could execute all those commands with the default `pi` user.

DieterReuter, 2016-07-28:
@marcosnils I just had the chance to check the latest standard Raspbian with user "pi". 
```
$ echo $PATH
/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
```
So, I think it's a good idea to include `/sbin`, `/usr/sbin` and `/usr/local/sbin` into the standard path of our `pirate` user, too.

Thanks for your finding.

DieterReuter, 2016-07-28:
Here is the current PATH of user `pirate` in HypriotOS:
```
$ echo $PATH
/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games
```

DieterReuter, 2016-07-28:
@marcosnils you can easily update your users `.profile`, so you don't have to wait for the next HypriotOS release. Login as `pirate` on your RPi and use this command:
```
curl -sSL -o ~/.profile https://github.com/hypriot/os-rootfs/raw/0451d86b2e3fa2ca3feac21894e7edc3d4288bc9/builder/files/etc/skel/.profile
```

StefanScherer, 2016-07-28:
Will be part of os-rootfs v0.8.7. we'll add this in another PR to this repo.

marcosnils, 2016-07-28:
> @marcosnils you can easily update your users .profile, so you don't have to wait for the next HypriotOS release. Login as pirate on your RPi and use this command:

yup, made the change as soon as I realized about it. Thx!.

StefanScherer, 2016-07-28:
Building v0.8.2 of the RPi SD card image. Should show up at GitHub releases later...

DieterReuter, 2016-07-28:
@marcosnils could you please test the new release v0.8.2, which contains the solutions of #107, #108 and #109. New release is available for download here https://github.com/hypriot/image-builder-rpi/releases/tag/v0.8.2. And please send feedback about using the camera module!

marcosnils, 2016-07-28:
@DieterReuter will try to do before the weekend.