hypriot/image-builder-rpi #119: Run root FS as read-only?

glynhudson, 2016-08-13:
Just discovered hypriot, nice work getting Docker to play nicely on a RasPi! :thumbsup: . 

I am I developer on the OpenEnergyMonitor Project, we currently use a RasPi3 with root FS in read-only for robust long-term operation. In testing we found SD cards hit their RW limit and failed in less then 12 months when used for logging application. 

My questions is can the hypriot SD card images work with root FS in read-only? Do you have any any experience / recommendations for running Docker on RaPi with read-only root FS? I have documented the [steps taken to make raspbian read-only here](https://github.com/emoncms/emoncms/blob/master/docs/RaspberryPi/read-only.md) and our [full SD card build guide here](https://github.com/openenergymonitor/emonpi/blob/master/docs/SD-card-build.md). 

I first posed this question on Gitter chat, @firecyberice suggested I open an issue here. [In a tweet](https://twitter.com/Quintus23M/status/763823757886619648) @DieterReuter  suggested this should be possible. Ram disks will be needed, we currently have the following setup

```
tmpfs           /tmp            tmpfs   nodev,nosuid,size=30M,mode=1777 0  0
tmpfs           /var/log        tmpfs   nodev,nosuid,size=50M,mode=1777 0  0
tmpfs           /var/lib/dhcp   tmpfs   nodev,nosuid,size=1M,mode=1777 0  0
tmpfs           /var/lib/openhab tmpfs  nodev,nosuid,size=40M,mode=1777 0  0
proc            /proc           proc    defaults 0 0
/dev/mmcblk0p1  /boot           vfat    defaults,noatime,nodiratime 0 2
/dev/mmcblk0p2  /               ext4    defaults,ro,noatime,nodiratime,errors=remount-ro 0 1
/dev/mmcblk0p3  /home/pi/data   ext2    defaults,rw,noatime,nodiratime,errors=remount-ro 0 2
```

`/home/pi/data` is mounted as RW then everything else is RO or tmpfs. Our [pre-built SD card images can be downloaded](https://github.com/openenergymonitor/emonpi/wiki/emonSD-pre-built-SD-card-Download-&-Change-Log) if you're interested how this works. We have had RaspberryPi's running without a hitch for other 3 years constantly data logging using this setup.

DieterReuter, 2016-08-14:
Thanks for this input, really interesting idea to run most of the OS in read-only mode. This leads to a longer lifetime for the SD card and should also increase the stability and security of the system. :+1:

I think we'll dig into this later. In the meantime you can try it on your own with our latest HypriotOS release - we're right now preparing a major upgrade, but it's too late to include new features.

StefanScherer, 2018-03-22:
Closing due to inactivity.

closed 2018-03-22 by StefanScherer