hypriot/image-builder-rpi #126: Prepare release notes for next HypriotOS v1.0.0 release "Blackbeard"

DieterReuter, 2016-08-14:
# HypriotOS 1.0.0-rc1

This is the first major release of HypriotOS for the Raspberry Pi - a Debian-based Container OS focussed on security and minimal size with all the necessary Docker tools already pre-installed. It is heavily inspired by Raspbian and supports absolutely all Raspberry Pi models (1, 2, 3, Zero & Compute Module) out-of-the-box. The Linux kernel is optimised for use of Docker containers and runs on both ARMv6 and ARMv7 CPU's.

**From Zero to Docker in five minutes only:**
Downloading the SD card image, flashing it to a SD card and booting up the Raspberry Pi takes less than five minutes - and the latest Docker Engine is already up and running.

## Contents
* Linux Kernel 4.4.15 (optimised for Docker)
  * dual-arch kernel for ARMv6 and ARMv7 (supports all Raspberry Pi models)
* Docker Engine 1.12.1-rc1 (officially build from Docker)
* Docker Machine 0.8.0
* Docker Compose 1.8.0
* Hypriot Device Init 0.1.8
* Hypriot OS-RootFS 1.0.0-rc1 - Raspbian/Jessie "Raspbian GNU/Linux 8 (jessie)"

## Features
* The root filesystem of the SD card will be automatically resized on first-boot
* No built-in "root" user for increased security
* Only a single standard user (username "pirate", password "hypriot") with `sudo` rights
  * User "pirate" is member of group "docker", can start `docker` commands directly
  * User "pirate" is member of group "video", can access the Raspberry Pi camera module
* `device-init` allows image customisation before first-boot
  * Standard hostname "black-pearl", can be configured in `/boot/device-init.yaml`
  * WiFi can be configured via `device-init` in `/boot/device-init.yaml`
  * [Hypriot Flash tool](https://github.com/hypriot/flash) directly supports `device-init` configuration
* Avahi service discovery built-in
  * find your Pi with a simple `ping black-pearl.local` or `ping black-pearl` on the network
* Console login via UART pins is enabled by default (for all Raspberry Pi models including Pi 3)
* almost all WiFi firmware included (atheros, brcm80211, libertas, ralink, realtek) for maximum compatibility
* Using Overlay filesystem for Docker storage-driver is activated by default

## Improvements
* Reduced Image size (1000M/1.0.0-rc1 vs. 1300M/v0.8.4) for quicker SD card flashing
  * flashing to a SanDisk Ultra 16GB microSD card takes 01:56min only (2014 MBP, OS X 10.11)
* Reduced .img.zip size (261M/1.0.0-rc1 vs. 504M/v0.8.4) for smaller and faster downloads
* Optimised filesystem to support more I-nodes (1x I-node per 4kByte block)
* Faster boot times (measured with a freshly flashed SD card, no containers, no additional apps)
  * first-boot: 21sec from power-on until to login prompt on UART console
  * normal boot: 15-17sec from power-on to login prompt on UART console
  * reboot: 22sec to login prompt on UART console
* Minimised memory usage, 46MByte only (new install, right after boot, w/ SSH session)
* Minimised disk usage, 780MByte only (new install)
```
$ df -h
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        14G  780M   13G   6% /
devtmpfs        427M     0  427M   0% /dev
tmpfs           431M     0  431M   0% /dev/shm
tmpfs           431M  5.7M  426M   2% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           431M     0  431M   0% /sys/fs/cgroup
/dev/mmcblk0p1   64M   30M   34M  47% /boot
```

## Removed Features
* Hypriot Cluster Lab (already available as separate .deb package)

## Known Issues
* this release currently only works on Pi 2 & Pi 3 as the Docker Engine 1.12.1-rc1 is built for ARMv7 only (should be fixed with the 1.0.0 release)

# LICENSE
The MIT License (MIT)

Copyright (c) 2016 Hypriot

MathiasRenner, 2016-08-18:
Here the updated release notes: 

# HypriotOS 1.0.0-rc1

This is the first major release of HypriotOS for the Raspberry Pi - a Debian-based Container OS focused on security and minimal size with all the necessary Docker tools already pre-installed. It is heavily inspired by Raspbian and supports absolutely all Raspberry Pi models (1, 2, 3, Zero & Compute Module) out-of-the-box. The Linux kernel is optimized for use of Docker containers and runs on both ARMv6 and ARMv7 CPU's.

**From Zero to Docker in five minutes only:**
Downloading the SD card image, flashing it to a SD card and booting up the Raspberry Pi takes less than five minutes - and the latest Docker Engine is already up and running.

## Contents
* Linux Kernel 4.4.15 (optimized for Docker)
  * dual-arch kernel for ARMv6 and ARMv7 (supports all Raspberry Pi models)
* Docker Engine 1.12.1-rc1 (officially build from Docker)
* Docker Machine 0.8.0
* Docker Compose 1.8.0
* [Hypriot device-init](https://github.com/hypriot/device-init) 0.1.8
* [HypriotOS-RootFS](https://github.com/hypriot/os-rootfs) 1.0.0-rc1 - Raspbian/Jessie "Raspbian GNU/Linux 8 (jessie)"

## Features
* The root file system of the SD card will be automatically resized on first boot
* No built-in "root" user for increased security
* Only a single standard user with `sudo` rights (username "pirate", password "hypriot")
  * User "pirate" is member of group "docker", can start `docker` commands directly
  * User "pirate" is member of group "video", can access the Raspberry Pi camera module
* `device-init` allows image customization before first boot
  * Standard hostname "black-pearl", can be configured in `/boot/device-init.yaml`
  * WiFi can be configured via `device-init` in `/boot/device-init.yaml`
  * [Hypriot Flash tool](https://github.com/hypriot/flash) directly supports `device-init` configuration
* Avahi service discovery built-in
  * find your Pi with a simple `ping black-pearl.local` or `ping black-pearl` on the network
* Console login via UART pins is enabled by default (for all Raspberry Pi models including Pi 3)
* Almost all WiFi firmware included for maximum compatibility (atheros, brcm80211, libertas, ralink, realtek)
* Using Overlay filesystem for Docker storage-driver is activated by default

## Improvements
* Reduced Image size for quicker SD card flashing (1000 MB/1.0.0-rc1 vs. 1300 MB/v0.8.4)
  * flashing to a SanDisk Ultra 16GB microSD card takes 01:56min only (2014 MBP, OS X 10.11)
* Reduced .img.zip size for smaller and faster downloads (252 MB/1.0.0-rc1 vs. 504 MB/v0.8.4)
* Optimized file system to support more Inodes (1x Inode per 4 KB block)
* Faster boot times (measured with a freshly flashed SD card, no containers, no additional apps)
  * first boot: 21 sec. from power-on to login prompt on UART console
  * normal boot: 15-17 sec. from power-on to login prompt on UART console
  * reboot: 22 sec. to login prompt on UART console
* Minimized memory usage, 46 MB only (new install, right after boot, w/ SSH session)
* Minimized disk usage, 750 MB only (new install)
```
$ df -h
**//TODO**
```

## Removed Features
* [Hypriot Cluster Lab](https://github.com/hypriot/cluster-lab) (already available as separate .deb package)

## Known Issues
* This release currently only works on Pi 2 & Pi 3 as the Docker Engine 1.12.1-rc1 is built for ARMv7 only (should be fixed with the 1.0.0 release)

# LICENSE
The MIT License (MIT)

Copyright (c) 2016 Hypriot

DieterReuter, 2016-08-18:
# HypriotOS 1.0.0-rc2

This is an intermediate release candidate with some small changes.

## Changes since 1.0.0-rc1
* Update os-rootfs to v1.0.0
* Don't install linux-headers to save disk space
* Update to official Docker Engine 1.12.1-rc2

## Known Issues
* this release currently only works on Pi 2 & Pi 3 as the Docker Engine 1.12.1-rc2 is built for ARMv7 only (should be fixed with the 1.0.0 release)

# LICENSE
The MIT License (MIT)

Copyright (c) 2016 Hypriot

DieterReuter, 2016-08-18:
# HypriotOS 1.0.0-rc3

This is an intermediate release candidate with some small changes.

## Changes since 1.0.0-rc3
* Update to official Docker Engine 1.12.1

## Resolved Issues
* this release works on all Raspberry Pi models (1, 2, 3, Zero & Compute Module) as the Docker Engine 1.12.1 is built for ARMv6

# LICENSE
The MIT License (MIT)

Copyright (c) 2016 Hypriot

DieterReuter, 2016-08-19:
Release completed!

🏁

closed 2016-08-19 by DieterReuter