hypriot/image-builder-rpi #127: Identify unnecessary packages

MathiasRenner, 2016-08-16:


MathiasRenner, 2016-08-18:
This is a comparison on packages between HypriotOS and Raspbian Lite
```
curl http://mathias-renner.de/dl/compare
```

On the left side you see packages that are unique in Raspbian Lite. On the right side the ones that are unique in HypriotOS.

After evaluating this, I suggest these categories:

Can be purged:
--------------------
  - **git-man (saves 1.3 MB)**: IMHO manpages can be removed. The ones on the workstation can be used in stead.
  - **git-core (saves 21.5 KB)**: Is a dummy package because  it has been renamed to just `git` (http://askubuntu.com/questions/5930/what-are-the-differences-between-the-git-and-git-core-packages)
  - **liberror-perl (saves 19 MB)** : Not used anywhere. Depends on `perl` which will then also be purged
  - **ethtool (saves 248 KB)**: It's installed in `chroot` script only to execute a test in `test-integration/spec/hypriotos-image/base/rpi3_wifi_spec.rb` 
  - **htop (saves 135 KB)**: Cool, but not necessary
  - **bluetooth**: This is a GUI that can be removed (http://www.cnet.com/how-to/how-to-setup-bluetooth-on-a-raspberry-pi-3/). Instead we should stick to `pi-bluetooth`. This listing shows the GUI package `obexd`:
```
$ apt-cache depends bluetooth
bluetooth
  Depends: bluez
  Suggests: <bluez-cups>
  Suggests: <bluez-obexd>

$ apt-cache depends pi-bluetooth
pi-bluetooth
  Depends: init-system-helpers
  Depends: bluez
  Depends: bluez-firmware
```

Can't be removed:
------------------
- **curl** is used to install Docker in `chroot-script.sh` and a dpenency of git (but git might be removed!)
- **aufs** is used by Docker
- **apt-transport-https** is required when adding our packagecloud packages
- **libapparmor1**: Required by Docker

Maybe remove:
-------------------
- **git (saves 19 MB)** is useful but maybe not for everyone

StefanScherer, 2018-03-22:
Closing due to inactivity.

closed 2018-03-22 by StefanScherer