hypriot/image-builder-rpi #131: Network boot with rpi 3

MichaelBitard, 2016-08-19:
It's not an issue so I apologize if I'm off-topic, but has anyone tested the network boot with the hypriot image?

This is quite new: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/net_tutorial.md and works only for the rpi 3.

I know that you just published the 1.0.0 blackbeard with docker 1.12.1 in it and I think it would be one step further to creating docker swarm cluster easily if we could network boot the rpis.

DieterReuter, 2016-08-19:
@MichaelBitard absolutely, would be awesome to include the Netboot feature in a future release. Thanks for sharing.

Right now, it's already under heavy development and we're waiting eagerly to get the final working feature from the Raspberry Pi Foundation released.

BTW, I've already tested Netboot and it looks great. I was even able to Netboot a Pi2 with some magic damn small SD card - you just need a single file on it. As soon as the final version is available, we'll work on including it into HypriotOS and will prepare a cool blog post - as you may guessing, and we'll run everything in Docker containers. 🐳

MichaelBitard, 2016-08-19:
That's great news, If by any mean I can help let me know.

closed 2016-08-19 by MichaelBitard