hypriot/image-builder-rpi #132: Add cgroup_enable=cpuset to /boot/cmdline.txt

luxas, 2016-08-28:
This is a Kubernetes requirement (hard dependency), ref: https://github.com/kubernetes/kubernetes/issues/26093

It would be great to have it enabled by default, then a reboot isn't necessary when installing Kubernetes on ARM :)

StefanScherer, 2016-08-29:
Indeed, this would be a great out-of-the-box experience :-) Can I persuade you to send a PR with the change in mind? The /boot/cmdline.txt is created in [builder/chroot-script.sh#L139](https://github.com/hypriot/image-builder-rpi/blob/a47caa671b834e6cd0afd755677ca6a224c76496/builder/chroot-script.sh#L139).
We could draft a pre-release to produce a SD card image to try it out with such a PR.

yonatan-schultz, 2016-09-02:
I think that something like this will do: https://github.com/hypriot/image-builder-rpi/pull/133

DieterReuter, 2016-09-02:
Fixed by #133

closed 2016-09-02 by DieterReuter

luxas, 2016-09-02:
Thanks @yonatan-schultz, was just about to send a PR, but now I have one thing less to do :)

@DieterReuter Any possibility an image with that patch could be produced so I can verify it's working OOTB? I'm releasing Kubernetes on ARM v0.8.0 really soon and it would be awesome with an Hypriot image with this cpuset by default :) (hopefully no reboot required)

yonatan-schultz, 2016-09-02:
@luxas no problem, thanks for all your work on Kubernetes. docker-multinode is making me very happy (once I found this fix)

DieterReuter, 2016-09-02:
@luxas I just tested it manually here on my Pi and the new integration tests as well. Will discuss with @StefanScherer to build a pre-release soon.

luxas, 2016-09-02:
Great, send an email with download info to me when you have it uploaded :+1:

DieterReuter, 2016-09-03:
@luxas I've just released a new HypriotOS 1.0.1 image for the Raspberry Pi, see https://github.com/hypriot/image-builder-rpi/releases/tag/v1.0.1.