hypriot/image-builder-rpi #143: same /var/lib/dbus/machine-id

StefanScherer, 2016-10-27:
Issued in hypriot/flash#92 by @cescoferraro, this seems a problem in building the image

> ```
> sudo -E flash -n athenas -s HACK-MEEE -p password https://downloads.hypriot.com/hypriotos-rpi-v1.0.0.img.zip
> 
> sudo -E flash -n esparta -s HACK-MEEE -p password https://downloads.hypriot.com/hypriotos-rpi-v1.0.0.img.zip
> 
> sudo -E flash -n zeus -s HACK-MEEE -p password https://downloads.hypriot.com/hypriotos-rpi-v1.0.0.img.zip
> 
> HypriotOS/armv7: pirate@athenas in ~
> $ cat /var/lib/dbus/machine-id
> 9989a26f06984d6dbadc01770f018e3b
> 
> HypriotOS/armv7: pirate@esparta in ~
> $ cat /var/lib/dbus/machine-id
> 9989a26f06984d6dbadc01770f018e3b
> 
> HypriotOS/armv6: pirate@zeus in ~
> $ cat /var/lib/dbus/machine-id
> 9989a26f06984d6dbadc01770f018e3b
> ```
> 
> On debian machines there is a hash under cat /var/lib/dbus/machine-id that uniquely identifies each machine. My program depends on it, but the flash tool seems to bring its own from https://downloads.hypriot.com/hypriotos-rpi-v1.0.0.img.zip. Is there another way I can uniquely identify a machine under Hypriot?

StefanScherer, 2018-03-22:
Fixed with #179

closed 2018-03-22 by StefanScherer

rhuss, 2018-03-22:
@StefanScherer cool, thx ! Do you know since version of Hypriot this fix is included so that I can update my README ?

StefanScherer, 2018-03-22:
Starting with HypriotOS 1.7.0 where we introduced cloud-init.
BTW: https://github.com/hypriot/image-builder-rpi/releases/tag/v1.8.0-rc1 pre-released today ;-) Would love to hear feedback how it works for you.

rhuss, 2018-03-22:
cool. I'm trying to give a spin later today ...