hypriot/image-builder-rpi #145: Could do with tips on how to run on your local Linux host

SvenDowideit, 2016-11-25:
I'd rather not need vagrant / another vm, and would like to build locally.

it failed with a pretty unhelpful error at the chroot in build.sh:

```
+ cp -R /builder/files/boot /builder/files/etc /build/
+ chroot /build /bin/bash
chroot: failed to run command '/bin/bash': No such file or directory
Makefile:7: recipe for target 'sd-image' failed
```

Its running now, but only after I installed 
```
apt-get install -y \
    binfmt-support \
    qemu \
    qemu-user-static
```

 on the ubuntu host I am on

StefanScherer, 2016-11-25:
Thanks for trying our build.  We never tried it directly on a Linux machine as you have to install all dependencies. Should we add something [like this](https://github.com/docker/docker/blob/master/hack/make.sh#L44-L53)? 😀

SvenDowideit, 2016-11-28:
really? wow :) IDK - it feels really weird to me to use a VM when I'm already on a Linux box - but then I've been using containers for rather too long.

fsamir, 2017-01-16:
It would be nice to document how to setup the building host, too. I have tried both on my mac and on an Ubuntu running on AWS, unsuccessfully because there is always a issue with either VirtualBox or vagrant or docker/docker-machine.
Thanks

StefanScherer, 2017-03-20:
As we build the SD image with CircleCI all needed steps are in circle.yml. Of course there are lots of preinstalled tools in the Circle builders.

StefanScherer, 2018-03-22:
We'll switch to Circle, so a .circleci/config.yml has all the commands you need to know how to build it on a Linux machine.

closed 2018-03-22 by StefanScherer