hypriot/image-builder-rpi #150: New user shell incompatible with default .bash_prompt

samthecodingman, 2017-01-08:
**Temporary Fix:**
For those who stumble across this and need a fix, follow these steps.

1. Login with the desired user
2. Run `chsh -s /bin/bash` to change the default shell to `/bin/bash`
3. Log out
4. Log in

**The Problem:**
The default `.bash_prompt` file created for new users is incompatible with `/bin/sh`. From what I can tell, this bash prompt file is used to prettify/color the console (named "@gf3's Sexy Bash Prompt"). But this results in syntax errors when the file is interpreted by `/bin/sh`. The new user needs to have the default shell changed to `/bin/bash` to handle this file correctly.

**Potential Fixes:**
Change the default shell of new users to `/bin/bash` rather than `/bin/sh` or remove the `.bash_prompt` 'prettifier' for new users.

**Other Notes:**
"But guy, why are you creating new users?" - Whenever I install a new OS, I change the default user so that it can't just be Googled for default login details. Normally I achieve this by adding a new user for SSH and then disable SSH entirely on the default `pirate` user (as it hosts the docker processes).

(P.S.: Awesome work folks. Keep it up. Also sorry in advance if I've made an error. Still relatively new to low-level Linux)

StefanScherer, 2018-03-22:
We have switch to cloud-init with HypriotOS 1.7.1.
You have much more control which users you want to have on first boot.
See also https://blog.hypriot.com/post/cloud-init-cloud-on-hypriot-x64/ for some examples.

StefanScherer, 2018-03-22:
Closing this as I assume this is fixed or can be fixed with a better user-data file.
If you still have problems please let me know.

Duvel, 2018-03-22:
Hi @StefanScherer ,

I don't think it should be solved with cloud-init. I encountered this problem when using cloud-init and had to set the default shell in the cloud-init script. This feels more as a conflict between settings.

Duvel, 2018-03-22:
@StefanScherer I see that you are working hard on a new release! Many thanks!

I will give the RC a try with my cloud-init and will report back if it works now as expected.

StefanScherer, 2018-03-22:
Ah thanks. I'm wiping the dust off some tickets and see what we can add to the next release ;-)

Duvel, 2018-03-22:
Just tested it with 1.8.0-RC1 and the default shell and .bashrc still don't match.

Setting the shell in user-data only works for your first boot. Users that are created afterwards still have this issue.

StefanScherer, 2018-03-22:
While looking for the wrong /bin/sh default, if you want to have a more secure Pi setup you could use cloud-init's `user-data` file before the first boot. We use this for our PiCloud where we have to flash a lot of SD cards. Look at the example in https://github.com/sealsystems/tiny-cloud/tree/master/prepare-hardware/pi#prepare-raspberry-pi-3 where you can use your SSH public key instead of a fixed password. So you don't have to create the user manually after the  boot.

StefanScherer, 2018-03-25:
Will be fixed with #228

closed 2018-03-25 by StefanScherer

StefanScherer, 2018-03-25:
Had to revert #228, `useradd` now created users without an shell entry in /etc/passwd which also defaults to /bin/sh.
As useradd is a lowlevel tool it's better to user `useradd newaccount --shell /bin/bash` or use `adduser`.

PR #228 (2018-03-25): Set useradd default shell to bash
`useradd abc` creates a new user but with `/bin/sh` as default shell. Set the default to `/bin/bash`

Fixes #150