hypriot/image-builder-rpi #151: Unable to to load dwc2 and g_ether modules

k4k, 2017-01-18:
The standard way of configuring the usb gadget networking is to set dwc2 and g_ether modules in cmdline.txt and setting "dtoverlay=dwc2" in config.txt. Doing this produces the following error under hypriot:

`Jan 18 16:42:06 master systemd-modules-load[116]: Failed to find module 'dwc2'
Jan 18 16:42:06 master systemd-modules-load[116]: Failed to find module 'g_ether'`

Would it be possible to have these modules included in the HypriotOS builds if they are not already? If they are, are they enabled in a manner different from Raspbian?

valentinvieriu, 2017-03-06:
This would be a very useful feature for the new RPI Zero w
[See Workflow](http://blog.gbaman.info/?p=791)

StefanScherer, 2018-03-22:
Please check our next HypriotOS release which we are working on in #220. A new kernel 4.9.80 is more identical to the Raspbian kernel. So there are good chances to have these kernel modules available.

StefanScherer, 2018-03-22:
First release candidate is available https://github.com/hypriot/image-builder-rpi/releases/tag/v1.8.0-rc1
Please help us and send us feedback if this fixes your problem.

closed 2019-09-16 by StefanScherer