hypriot/image-builder-rpi #157: adding support for cloud-init

mwerner-tp, 2017-01-28:
Using device-init.yaml https://github.com/hypriot/device-init is a great for a easy setup and is really usefull for entrylevel, since it is not to complex. 

Anyway cloud-init gives you more flexibility in what you configure. So it would be got to have hypriotOS to support it.

DieterReuter, 2017-01-28:
@mwerner-tp sounds like a great idea. Could you please dig into it and create a PR with a first working version, please? Then we can use this PR as a discussion and hopefully we can finally integrate it into HypriotOS - I'd love to support this too.

mwerner-tp, 2017-01-29:
@DieterReuter, my dev skills are not good enough to accomplish this task. I don't know where to start at the first place.

StefanScherer, 2017-02-20:
@mwerner-tp A proof-of-concept can be found at https://github.com/sealsystems/pi-cloud/blob/master/Setup.md#hypriotos

gesellix, 2017-03-22:
I can confirm that the linked proof-of-concept works nicely (updated link: https://github.com/sealsystems/tiny-cloud/tree/f31dd0f8db848eada7b49d40620037a4f6d86adf/prepare-hardware/pi).

StefanScherer, 2017-03-22:
@gesellix Thanks for the feedback.

A question to all: What default should be used if we introduce cloud-init?

- [ ] No `/boot/user-data` at all or only a disabled `/boot/user-data.sample` 
- [ ] A `/boot/user-data` with user `pirate` and password `hypriot` to have old behaviour, but less secure than standard Raspbian without SSH. They have learned the lesson of users not changing default password and putting RPi's online.

gesellix, 2017-03-22:
What happens in the first scenario when the user forgets to configure a user? Will it be possible to show a helpful error message?
I tend to vote for the first option (with a disabled `user-date.sample` user), exactly due to security reasons like you already mentioned.

burdiyan, 2017-06-19:
Is there any way to use cloud-init right now? I see https://github.com/hypriot/flash has option to provide cloud-init user data, but there is no cloud-init binary in the default Hypriot image.

StefanScherer, 2017-06-19:
@burdiyan At the moment we haven't decided yet when to switch over to cloud-init as we then have to remove device-init, but you can use my fork right now https://github.com/StefanScherer/image-builder-rpi/releases/tag/v1.4.0 and give us feedback.

burdiyan, 2017-06-19:
@StefanScherer maybe a healthy solution can be keep both cloud-unit and device-init ina way that they are mutually exclusive. So if you flash the image with user data flag device-int is ignored and vice versa.

WDYT?

StefanScherer, 2017-06-19:
@burdiyan Yes that might be a way to migrate. The flash script should remove the default device-init.yaml if someone specifies a user-data.

StefanScherer, 2017-11-01:
Implemented with #192

closed 2017-11-01 by StefanScherer