hypriot/image-builder-rpi #159: static ip address

tsouhaieb, 2017-02-15:
how to fix ip address, on raspbian jessy i normally edit /etc/dhcpcd.conf file,
but in hypriot os this file doesn't exist and even if i create it and edit i, the change in it does not take effect even after reboot, how we can fix and change fixed ip address on hypriotos, sorry for the bothering and thanks for this great work,

Salam,
Peace

carljmosca, 2017-02-15:
check /etc/network/interfaces.d for your interface(s)

firecyberice, 2017-02-16:
Currently we do not have dhcpcd installed by default so creating `/etc/dhcpcd.conf` has no effect.

tsouhaieb, 2017-02-16:
i already find a way, you can check [device-init#6](https://github.com/hypriot/device-init/issues/6)

closed 2017-02-16 by tsouhaieb

StefanScherer, 2018-03-25:
We have switched to cloud-init. 
Cloud-init normally does static IP with other files, but then you need the MAC address of your Pi. But what if you want to unbox a fresh RPi and flash and boot it?

With cloud-init there is another hacky way to write the information into `/boot/user-data` file

```yaml
#cloud-config
# vim: syntax=yaml
#

# Set your hostname here, the manage_etc_hosts will update the hosts file entries as well
hostname: static
manage_etc_hosts: true
# don't write debian.org into apt mirrors
apt_preserve_sources_list: true

# You could modify this for your own user information
users:
  - name: pirate
    gecos: "Hypriot Pirate"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video
    plain_text_passwd: hypriot
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }

package_upgrade: false

# Static IP address
write_files:
  - content: |
      persistent
      # Generate Stable Private IPv6 Addresses instead of hardware based ones
      slaac private

      # static IP configuration:
      interface eth0
      static ip_address=192.168.1.31/24
      #static ip6_address=fd51:42f8:caae:d92e::ff/64
      static routers=192.168.1.1
      static domain_name_servers=192.168.1.1 8.8.8.8

    path: /etc/dhcpcd.conf
```

machine1104, 2019-02-19:
not working for me :(

isc30, 2019-03-19:
same, not working with v1.10.0

vispatel, 2020-05-31:
not working for me either