hypriot/image-builder-rpi #167: /etc/machine-id the same on every installation

rhuss, 2017-04-04:
First of all apologies if this is the wrong place for opening this issue (feel free to move it around)

I hit the following issue that on all my pi installations with Hypriot 1.4.0 I get the same `/etc/machine-id`. This causes [issues](https://github.com/luxas/kubeadm-workshop/issues/7) when running an overlay network which uses this ID to calculate a virtual mac address.

Actually the _same id_ 9989a26f06984d6dbadc01770f018e3b remains even after a *reflash*. And even stranger, when removing this file and calling `systemd-machine-id-setup` it recreates the id file with again this value. Looks for me a bit like an issue with how systemd extracts an UUID (it says "Initializing machine ID from D-Bus machine ID.", so maybe that's a constant ?)

The solution for now is that after flashing I call a 

```
dbus-uuidgen > /etc/machine-id
```

or when using ansible with an ad-hoc ansible command.

```
ansible pis -u pirate -k -i hosts --become -m shell --args "dbus-uuidgen > /etc/machine-id"
```

Maybe one could add something similar to the initial startup of a freshly flashed Hypriot ?

StefanScherer, 2017-04-04:
Same as #143, but thanks for more details.

rhuss, 2017-04-04:
thanks and sorry for not checking the other issues first.

closed 2017-07-23 by firecyberice

StefanScherer, 2018-03-22:
Fixed with #179