hypriot/image-builder-rpi #170: Avahi demon dead after startup, service needs to be restarted manually

maxromanovsky, 2017-04-18:
After startup on the newest `v1.4.0` `.local` DNS does not work.
`systemctl start avahi-daemon.service` have to be executed manually in order to start it up.

```bash
HypriotOS/armv7: root@pi-k8s-node02 in ~
$ systemctl status avahi-daemon.service -l
● avahi-daemon.service - Avahi mDNS/DNS-SD Stack
   Loaded: loaded (/lib/systemd/system/avahi-daemon.service; enabled)
   Active: inactive (dead) since Tue 2017-04-18 08:27:09 UTC; 9h ago
  Process: 402 ExecStart=/usr/sbin/avahi-daemon -s (code=exited, status=0/SUCCESS)
 Main PID: 402 (code=exited, status=0/SUCCESS)
   Status: "avahi-daemon 0.6.31 starting up."
```

Logs before manual `start`:
```bash
HypriotOS/armv7: root@pi-k8s-node02 in ~
$ journalctl --no-pager -u avahi-daemon.service
-- Logs begin at Tue 2017-04-18 18:33:36 UTC, end at Tue 2017-04-18 18:35:57 UTC. --
Apr 18 18:33:39 pi-k8s-node02 systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Apr 18 18:33:39 pi-k8s-node02 avahi-daemon[354]: Found user 'avahi' (UID 105) and group 'avahi' (GID 111).
Apr 18 18:33:39 pi-k8s-node02 avahi-daemon[354]: Successfully dropped root privileges.
Apr 18 18:33:39 pi-k8s-node02 avahi-daemon[354]: avahi-daemon 0.6.31 starting up.
Apr 18 18:33:39 pi-k8s-node02 avahi-daemon[354]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Apr 18 18:33:39 pi-k8s-node02 avahi-daemon[354]: Successfully called chroot().
Apr 18 18:33:39 pi-k8s-node02 avahi-daemon[354]: Successfully dropped remaining capabilities.
Apr 18 18:33:39 pi-k8s-node02 avahi-daemon[354]: No service file found in /etc/avahi/services.
Apr 18 18:33:39 pi-k8s-node02 avahi-daemon[354]: Network interface enumeration completed.
Apr 18 18:33:39 pi-k8s-node02 avahi-daemon[354]: Registering HINFO record with values 'ARMV7L'/'LINUX'.
Apr 18 18:33:40 pi-k8s-node02 avahi-daemon[354]: Server startup complete. Host name is pi-k8s-node02.local. Local service cookie is 3761524834.
Apr 18 18:33:40 pi-k8s-node02 systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Apr 18 18:33:40 pi-k8s-node02 systemd[1]: Stopping Avahi mDNS/DNS-SD Stack...
Apr 18 18:33:40 pi-k8s-node02 avahi-daemon[354]: Got SIGTERM, quitting.
Apr 18 18:33:41 pi-k8s-node02 systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Process 354 died: No such process; trying to remove PID file. (/var/run/avahi-daemon//pid)
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Found user 'avahi' (UID 105) and group 'avahi' (GID 111).
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Successfully dropped root privileges.
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: avahi-daemon 0.6.31 starting up.
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Successfully called chroot().
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Successfully dropped remaining capabilities.
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: No service file found in /etc/avahi/services.
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Joining mDNS multicast group on interface eth0.IPv6 with address fe80::ba27:ebff:fe12:f42a.
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: New relevant interface eth0.IPv6 for mDNS.
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Network interface enumeration completed.
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Registering new address record for fe80::ba27:ebff:fe12:f42a on eth0.*.
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Registering HINFO record with values 'ARMV7L'/'LINUX'.
Apr 18 18:33:41 pi-k8s-node02 systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Apr 18 18:33:41 pi-k8s-node02 avahi-daemon[402]: Server startup complete. Host name is pi-k8s-node02.local. Local service cookie is 2618832816.
Apr 18 18:33:42 pi-k8s-node02 avahi-daemon[402]: Leaving mDNS multicast group on interface eth0.IPv6 with address fe80::ba27:ebff:fe12:f42a.
Apr 18 18:33:42 pi-k8s-node02 avahi-daemon[402]: Joining mDNS multicast group on interface eth0.IPv6 with address 2a02:bf0:1102:5f00:ba27:ebff:fe12:f42a.
Apr 18 18:33:42 pi-k8s-node02 avahi-daemon[402]: Registering new address record for 2a02:bf0:1102:5f00:ba27:ebff:fe12:f42a on eth0.*.
Apr 18 18:33:42 pi-k8s-node02 avahi-daemon[402]: Withdrawing address record for fe80::ba27:ebff:fe12:f42a on eth0.
Apr 18 18:33:45 pi-k8s-node02 avahi-daemon[402]: Joining mDNS multicast group on interface docker0.IPv4 with address 172.17.0.1.
Apr 18 18:33:45 pi-k8s-node02 avahi-daemon[402]: New relevant interface docker0.IPv4 for mDNS.
Apr 18 18:33:45 pi-k8s-node02 avahi-daemon[402]: Registering new address record for 172.17.0.1 on docker0.IPv4.
Apr 18 18:33:48 pi-k8s-node02 avahi-daemon[402]: Joining mDNS multicast group on interface eth0.IPv4 with address 192.168.0.106.
Apr 18 18:33:48 pi-k8s-node02 avahi-daemon[402]: New relevant interface eth0.IPv4 for mDNS.
Apr 18 18:33:48 pi-k8s-node02 avahi-daemon[402]: Registering new address record for 192.168.0.106 on eth0.IPv4.
Apr 18 18:33:48 pi-k8s-node02 systemd[1]: Stopping Avahi mDNS/DNS-SD Stack...
Apr 18 18:33:48 pi-k8s-node02 avahi-daemon[402]: Got SIGTERM, quitting.
Apr 18 18:33:48 pi-k8s-node02 avahi-daemon[402]: Leaving mDNS multicast group on interface docker0.IPv4 with address 172.17.0.1.
Apr 18 18:33:48 pi-k8s-node02 avahi-daemon[402]: Leaving mDNS multicast group on interface eth0.IPv6 with address 2a02:bf0:1102:5f00:ba27:ebff:fe12:f42a.
Apr 18 18:33:48 pi-k8s-node02 systemd[1]: Stopped Avahi mDNS/DNS-SD Stack.
```

And after:
```bash
Apr 18 18:35:56 pi-k8s-node02 systemd[1]: Starting Avahi mDNS/DNS-SD Stack...
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Process 402 died: No such process; trying to remove PID file. (/var/run/avahi-daemon//pid)
[... 16 lines cut ...]
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: New relevant interface flannel.1.IPv6 for mDNS.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Joining mDNS multicast group on interface flannel.1.IPv4 with address 10.244.1.0.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: New relevant interface flannel.1.IPv4 for mDNS.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Joining mDNS multicast group on interface docker0.IPv4 with address 172.17.0.1.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: New relevant interface docker0.IPv4 for mDNS.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Joining mDNS multicast group on interface eth0.IPv6 with address 2a02:bf0:1102:5f00:ba27:ebff:fe12:f42a.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: New relevant interface eth0.IPv6 for mDNS.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Joining mDNS multicast group on interface eth0.IPv4 with address 192.168.0.106.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: New relevant interface eth0.IPv4 for mDNS.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Network interface enumeration completed.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering new address record for fe80::fcc7:3aff:feb6:2b8e on veth18a1bfd4.*.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering new address record for fe80::9043:2cff:fe02:bdf0 on veth32d68a48.*.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering new address record for fe80::3872:c9ff:feec:6692 on cni0.*.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering new address record for 10.244.1.1 on cni0.IPv4.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering new address record for fe80::48f0:c1ff:fec5:3a2a on flannel.1.*.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering new address record for 10.244.1.0 on flannel.1.IPv4.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering new address record for 172.17.0.1 on docker0.IPv4.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering new address record for 2a02:bf0:1102:5f00:ba27:ebff:fe12:f42a on eth0.*.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering new address record for 192.168.0.106 on eth0.IPv4.
Apr 18 18:35:56 pi-k8s-node02 avahi-daemon[2115]: Registering HINFO record with values 'ARMV7L'/'LINUX'.
Apr 18 18:35:56 pi-k8s-node02 systemd[1]: Started Avahi mDNS/DNS-SD Stack.
Apr 18 18:35:57 pi-k8s-node02 avahi-daemon[2115]: Server startup complete. Host name is pi-k8s-node02.local. Local service cookie is 3223266484
```

Sorry, I don't know the proper fix, so no PR for this one.

jonnor, 2018-10-03:
Where you able to find a fix? Seeing the same issue here.

StefanScherer, 2019-09-16:
Closing due to inactivity

closed 2019-09-16 by StefanScherer

everdrone, 2021-06-08:
I'm having the same issue with avahi-daemon, anybody found a solution?