hypriot/image-builder-rpi #172: How to create a ready to use USB image

gregorkistler, 2017-05-13:
Hi,

is there a way to flash the SD card image to an USB memory stick and boot it completely without a SD card?
Sadly the `cmdline.txt` and `/etc/fstab` contain block device names like `/dev/mmcblk0p2` for the root partition and not PARTUUIDs - same goes for the boot partition.
You´d need to boot from SD card first, copy everything to an USB memory stick and change both files - theoretically.

BUT:
Even if I try to get it done like described [here](https://raspberrypi.stackexchange.com/questions/65838/raspbian-on-usb-stick/65846) or [here](http://www.makeuseof.com/tag/make-raspberry-pi-3-boot-usb/) it still won´t boot completely.
My Raspberry Pi 3 gets an IP address and the network device responds to ICMP packages but the boot process always got stuck at a certain point.

Thanks

gregorkistler, 2017-05-13:
Got it. The only file which needs to be changed is `builder/chroot-script.sh`.
Created a new image and flashed it directly onto an USB memory stick.
Boots up without any issues.

Still not sure why it doesn´t work properly if I rsync the data from a working SD image installation though.

DieterReuter, 2017-05-13:
@gregorkistler Great! Could you please share the changes you made, please? Do you think we could include in a way, so one could boot HypriotOS from SD card, USB drive or Network boot with minimal chances only?

gregorkistler, 2017-05-14:
It´s actually just a quick'n'dirty solution as it would break the usage of SD cards and also works only as long as you just use one USB memory stick so I won´t recommend a PR.

The following changes were necessary:

1. `builder/chroot-script.sh`
Line 142: replace `mmcblk0p2` with `sda2`
Line 172: replace `mmcblk0p1` with `sda1`
Line 173: replace `mmcblk0p2` with `sda2`

2. `builder/files/etc/firstboot.d/10-resize-rootdisk`
Line 10: replace `mmcblk0p` with `sda`
Line 23: replace `mmcblk0` with `sda`
Line 31: replace `mmcblk0` with `sda`
Line 38: replace `mmcblk0` with `sda`

closed 2017-08-06 by gregorkistler