hypriot/image-builder-rpi #174: Built-in Bluetooth does not start on a pi zero w

bcopy, 2017-05-27:
Probably related to #72 
Test case :
* Hypriot image : 4.4.50-hypriotos+ #2 PREEMPT Sun Mar 19 14:44:01 UTC 2017 armv6l GNU/Linux

* The hciuart service fails to startup : 
```
$ sudo systemctl status hciuart.service -l
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/etc/systemd/system/hciuart.service; enabled)
   Active: failed (Result: exit-code) since Sat 2017-05-27 09:49:41 UTC; 53min ago
  Process: 363 ExecStart=/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow - (code=exited, status=1/FAILURE)

May 27 09:49:41 groove-assistant systemd[1]: hciuart.service: control process exited, code=exited status=1
May 27 09:49:41 groove-assistant systemd[1]: Failed to start Configure Bluetooth Modems connected by UART.
May 27 09:49:41 groove-assistant hciattach[363]: Initialization timed out.
May 27 09:49:41 groove-assistant hciattach[363]: bcm43xx_init
May 27 09:49:41 groove-assistant systemd[1]: Unit hciuart.service entered failed state.
```
* The file hciuart.service refers to serial1:
```
$ more /lib/systemd/system/hciuart.service
[Unit]
Description=Configure Bluetooth Modems connected by UART
ConditionPathIsDirectory=/proc/device-tree/soc/gpio@7e200000/bt_pins
Before=bluetooth.service
After=dev-serial1.device

[Service]
Type=forking
ExecStart=/usr/bin/btuart

[Install]
WantedBy=multi-user.target
```

* Devices  /dev/serial1 and /dev/ttyAMA0 both exist.

cnadeau, 2017-08-23:
looks like the kernel module hci_uart is not present and not compiled for the right architecture on the pi zero w armv6l (compared to the pi3 armv7l). Is there a compiled version for armv6l anywhere?

StefanScherer, 2017-08-24:
Please check our FAQ how to enable onboard WiFi in the boot settings. I guess this also helps for Bluetooth as both use UART.

cnadeau, 2017-08-25:
Thanks for the reply @StefanScherer! 

I missed that I could avoid the step changing it manually. Thanks! 

But it did not help from a fresh 1.5 image.

I flashed it, tried 
```
$ hcitools dev
Device:
``` 

I `apt-get upgrade`, 
same thing with or without a reboot

I'm not kernel driver expert, but shouldn't the hci_uart driver be loaded in order to get bluetooth to work?

And shouldn't `hci_uart.ko` be compiled for and present in `/lib/modules/4.4.50-hypriotos+/kernel/drivers/bluetooth`? 

It's only available in `/lib/modules/4.4.50-hypriotos-v7+/kernel/drivers/bluetooth`

OR, if it's not required, maybe this patch [#78](https://github.com/hypriot/image-builder-rpi/pull/78/files) is not right for the Pi Zero W?

I've found a couple of posts, but nothing that leads me to a fully functional BLE chip from hypriot.

Thanks a lot for you help!

StefanScherer, 2017-08-29:
Oh, it seems there is missing something. I don't have a use-case for bluetooth, so I can't put much time into it. Maybe using standard Raspbian and Docker works for you better.

bcopy, 2017-10-01:
Hi, could there be a list of known Raspbian functionalities and drivers known NOT to work with hypriot ? We're talking bluetooth now but what else doesn't work in hypriot ? it's fine to use Raspbian instead, but no one should spend time trying to make something known to be unsupported work.

TomCools, 2017-10-22:
Is this ever going to get resolved, or should i just switch to Raspbian?

StefanScherer, 2017-12-19:
Seems like this kernel config is missing for RPi 0/1, but I cannot test it.

[image]

I've opened hypriot/rpi-kernel#37

StefanScherer, 2018-03-22:
Good news, our new kernel 4.9.80 looks promising:

```
$ sudo systemctl status hciuart.service -l
● hciuart.service - Configure Bluetooth Modems connected by UART
   Loaded: loaded (/etc/systemd/system/hciuart.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-03-22 06:18:25 UTC; 8min ago
  Process: 481 ExecStart=/usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow - (code=exited, status=0/SUCCESS)
 Main PID: 777 (hciattach)
    Tasks: 1 (limit: 4915)
   Memory: 576.0K
      CPU: 40ms
   CGroup: /system.slice/hciuart.service
           └─777 /usr/bin/hciattach /dev/ttyAMA0 bcm43xx 921600 noflow -

Mar 22 06:18:21 circle21 systemd[1]: Starting Configure Bluetooth Modems connected by UART...
Mar 22 06:18:25 circle21 hciattach[481]: bcm43xx_init
Mar 22 06:18:25 circle21 hciattach[481]: Flash firmware /lib/firmware/brcm/BCM43430A1.hcd
Mar 22 06:18:25 circle21 hciattach[481]: Set Controller UART speed to 921600 bit/s
Mar 22 06:18:25 circle21 hciattach[481]: Device setup complete
Mar 22 06:18:25 circle21 systemd[1]: Started Configure Bluetooth Modems connected by UART.
```

StefanScherer, 2018-03-22:
Will be fixed with the next HypriotOS release we're working on in #220.

StefanScherer, 2018-03-22:
First release candidate is available https://github.com/hypriot/image-builder-rpi/releases/tag/v1.8.0-rc1

closed 2018-03-22 by StefanScherer