hypriot/image-builder-rpi #180: ads7846 touch screen

adgaudio, 2017-08-27:
Hello,

Hypriot OS is awesome!  Thank you for providing this OS - docker is great on it!  

I have a touch screen issue, and I would love your guidance in solving it.

I have a 5" touch screen that works just fine on Raspbian.  When I switch to Hypriot OS, the display works but touch does not.  Looking into this, I found that the `ads7846` module, which I believe is required for my touch screen to work, is not loaded.  The display is the "xpt2046 Touch Controller".  See the following log of commands I run on my pi running Hypriot OS:

```
$ cat /etc/issue
HypriotOS (Debian GNU/Linux 8) \n \l
$ uname -a
Linux makerspace-access-control 4.4.50-hypriotos-v7+ #1 SMP PREEMPT Sun Mar 19 14:11:54 UTC 2017 armv7l GNU/Linux
```
```
$ sudo lsmod|grep ads || echo did not find module
did not find module
$ cat /proc/interrupts |grep ads || echo module should be listed in interrupts
module should be listed in interrupts
$ $ cat /boot/config.txt |grep -E 'ads78|dtpar|hdmi|max_curr'|grep -v '^#'
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0
hdmi_drive=2
dtparam=i2c_arm=on
dtparam=spi=on
dtoverlay=ads7846,penirq=25,speed=10000,penirq_pull=2,xohms=150
dtparam=audio=on
```
```
# attempt manually mounting the overlay
$ sudo dtoverlay -v ads7846 penirq=25 speed=10000 penirq_pull=2 xohms=150
run_cmd: mount -t configfs none '/config/device-tree'
mount: mount point /config/device-tree does not exist
* Failed to mount configfs - 2
```

Clearly, the issue here is that the dtoverlay module does not load, neither on startup nor manually.

I'm not sure what to do, and I don't really understand device-tree configuration that well.  Do you have any suggestions for me?

Thank you,
Alex

adgaudio, 2017-09-11:
As an update on this, the touch screen is not functional on the official raspberry pi touch screen, as well.  Since it seems that HypriotOS does not support touch screens, I am no longer trying to use this OS and have switched to Raspbian.  

If you have any ideas on how to move forward, I'd be more than happy to work with you on this issue!

StefanScherer, 2017-09-11:
It would be difficult and time consuming for us to support all kinds of hardware. Yes, dt support is very minimal. So it's ok to go with Raspbian for such hardware.

adgaudio, 2017-09-11:
Ok, thank you for letting me know, Stefan.

closed 2017-09-11 by adgaudio