hypriot/image-builder-rpi #187: plans to move to 4.9.x kernel?

jgangemi, 2017-09-28:
are there plans to move to the `4.9.x` kernel and if so, what is the time frame?  the overlay for my `justboom` amp hat isn't being loaded properly by the current kernel.

thanks!

firecyberice, 2017-09-28:
Please have a look at [this](https://github.com/hypriot/rpi-kernel) repo for kernel upgrades. We are happy about PRs. Currently there is no time frame for an upgrade.

jgangemi, 2017-09-30:
ok - i am attempting this. 

do i need to run `menuconfig` as part of creating the PR or just recompile the kernel? do i just `exit` when i'm dropped into it? do i need to `save` first?

jgangemi, 2017-09-30:
also - why does the compile `rm -rf` everything instead of just updating an existing clone? makes re-running this somewhat painful.

firecyberice, 2017-09-30:
Which `rm -rf` do you mean? Please provide a link to the corresponding lines.  In general It is more save to start every build with a clean environment to minimize side effects.

jgangemi, 2017-10-01:
i switched to building on some aws hardare which made things a lot better.

unfortunately, i think this may be beyond me. after changing the commit hash and running the compile script, i get prompted to answer a bunch of questions for the `.config` file for new settings that i just accept all the defaults for (except when it asks about `arm6` support, i enable that) and then the build fails w/ this
```
ERROR: "rpi_firmware_get" [drivers/video/backlight/rpi_backlight.ko] undefined!
ERROR: "rpi_firmware_property" [drivers/video/backlight/rpi_backlight.ko] undefined!
scripts/Makefile.modpost:91: recipe for target '__modpost' failed
make[3]: *** [__modpost] Error 1
Makefile:1199: recipe for target 'modules' failed
make[2]: *** [modules] Error 2
scripts/package/Makefile:91: recipe for target 'deb-pkg' failed
make[1]: *** [deb-pkg] Error 2
Makefile:1332: recipe for target 'deb-pkg' failed
make: *** [deb-pkg] Error 2
```
so i'm guessing somewhere one or more of the default values is not correct for the build.

for now i'll just have to stick w/ the raspbian install but i'll def switch back once you and my amp hat will work.

ngpitt, 2017-11-09:
Not sure if this helps, but I'm successfully using k8s 1.8.3 with latest kernel (1.20171029-1) from archive.raspberrypi.org (stretch) with docker-ce (17.09.0~ce-0~raspbian). Replace all "cgroup_enable" args with "cgroup_memory=1" and all preflight checks will pass.

DieterReuter, 2017-11-09:
@ngpitt cool, thanks for the tip!

StefanScherer, 2018-03-22:
We're working on in #220

StefanScherer, 2018-03-22:
First release candidate is available https://github.com/hypriot/image-builder-rpi/releases/tag/v1.8.0-rc1

closed 2018-03-22 by StefanScherer