hypriot/image-builder-rpi #189: Unable to get the DS18B20 Temperature Sensor working on hypriot image

hreidar, 2017-10-04:
Hi, I using your image for a project, but I'm unable to get the 1-wire sensor to work. The same sensor works if I swap to the latest raspian image. Any ideas?

I'm following these instructions -> [https://matthewmoisen.com/blog/how-to-use-the-ds18b20-temperature-sensor-with-raspberry-pi/](url)

StefanScherer, 2017-10-05:
I haven‘t used 1-wire yet. So it‘s had to say, but probably some different kernel modules compiled in. 
I suggest to stay with Raspbian LITE and run

```
curl https://get.docker.com | sh
```

to install Docker 17.05 (17.09 is around the corner with the same command in a few days). 

As long as you do not want to run a Docker Swarm mode, then this would be better for such hardware projects.

hreidar, 2017-10-06:
Thanks for your reply. I'm sad that 1-wire is not working on hypriot image because it's awesome and I really don't want to use anything else :-)

It seems to me that the kernel modules are there as I can run the modprobe commands without an error and lsmod also does list them. I hope someone clever can find time to look into this matter, meanwhile, I will be forced to battle with Raspbian.

hreidar, 2017-11-11:
Hi all, just wanted to point out another OS for Raspberry whitch supports headless configuration, docker and 1-wire perfectly. [https://resinos.io/](url)

closed 2019-09-16 by StefanScherer