hypriot/image-builder-rpi #190: Running docker with TCP port binding fails

rhuss, 2017-10-14:
I want to enable the Docker daemon to listen on external ports so that they can be managed from the outside. However, during startup the Docker daemon fails to bind to the port.

To prove the point, the following simplified setup listens on the unprotected on port 2375.

* Edit `/lib/systemd/system/docker.service` and remove all options after `ExecStart`:

```
...
[Service]
Type=notify
# the default is not to use systemd for cgroups because the delegate issues still
# exists and systemd currently does not support the cgroup feature set required
# for containers run by docker
ExecStart=/usr/bin/dockerd
....
```

* Add a `/etd/docker/daemon.json` that looks like (with 192.168.23.200 being the external IP)

```
{
  "hosts": [ "fd://", "tcp://192.168.23.203:2375" ]
}
```

* Reboot 
* Docker daemon does not startup with the following error message:

```
# journalctl -l -u docker
-- Logs begin at Sat 2017-10-14 07:00:06 UTC, end at Sat 2017-10-14 07:01:22 UTC. --
Oct 14 07:00:20 n3 systemd[1]: Starting Docker Application Container Engine...
Oct 14 07:00:21 n3 dockerd[452]: time="2017-10-14T07:00:21.995502337Z" level=warning msg="[!] DON'T BIND ON ANY IP ADDRESS WITHOUT setting --tlsverify IF YOU DON'T KNOW WHAT YOU'RE DOING [!]"
Oct 14 07:00:21 n3 dockerd[452]: listen tcp 192.168.23.203:2375: bind: cannot assign requested address
Oct 14 07:00:22 n3 systemd[1]: docker.service: main process exited, code=exited, status=1/FAILURE
Oct 14 07:00:22 n3 systemd[1]: Failed to start Docker Application Container Engine.
Oct 14 07:00:22 n3 systemd[1]: Unit docker.service entered failed state.
Oct 14 07:00:22 n3 systemd[1]: docker.service holdoff time over, scheduling restart.
Oct 14 07:00:22 n3 systemd[1]: Stopping Docker Application Container Engine...
Oct 14 07:00:22 n3 systemd[1]: Starting Docker Application Container Engine...
Oct 14 07:00:22 n3 dockerd[506]: time="2017-10-14T07:00:22.683000514Z" level=warning msg="[!] DON'T BIND ON ANY IP ADDRESS WITHOUT setting --tlsverify IF YOU DON'T KNOW WHAT YOU'RE DOING [!]"
Oct 14 07:00:22 n3 dockerd[506]: listen tcp 192.168.23.203:2375: bind: cannot assign requested address
Oct 14 07:00:22 n3 systemd[1]: docker.service: main process exited, code=exited, status=1/FAILURE
Oct 14 07:00:22 n3 systemd[1]: Failed to start Docker Application Container Engine.
Oct 14 07:00:22 n3 systemd[1]: Unit docker.service entered failed state.
Oct 14 07:00:23 n3 systemd[1]: docker.service holdoff time over, scheduling restart.
Oct 14 07:00:23 n3 systemd[1]: Stopping Docker Application Container Engine...
Oct 14 07:00:23 n3 systemd[1]: Starting Docker Application Container Engine...
Oct 14 07:00:23 n3 dockerd[530]: time="2017-10-14T07:00:23.404669993Z" level=warning msg="[!] DON'T BIND ON ANY IP ADDRESS WITHOUT setting --tlsverify IF YOU DON'T KNOW WHAT YOU'RE DOING [!]"
Oct 14 07:00:23 n3 dockerd[530]: listen tcp 192.168.23.203:2375: bind: cannot assign requested address
Oct 14 07:00:23 n3 systemd[1]: docker.service: main process exited, code=exited, status=1/FAILURE
Oct 14 07:00:23 n3 systemd[1]: Failed to start Docker Application Container Engine.
Oct 14 07:00:23 n3 systemd[1]: Unit docker.service entered failed state.
Oct 14 07:00:23 n3 systemd[1]: docker.service holdoff time over, scheduling restart.
Oct 14 07:00:23 n3 systemd[1]: Stopping Docker Application Container Engine...
Oct 14 07:00:23 n3 systemd[1]: Starting Docker Application Container Engine...
Oct 14 07:00:23 n3 systemd[1]: docker.service start request repeated too quickly, refusing to start.
Oct 14 07:00:23 n3 systemd[1]: Failed to start Docker Application Container Engine.
Oct 14 07:00:23 n3 systemd[1]: Unit docker.service entered failed state.
Oct 14 07:01:11 n3 systemd[1]: Starting Docker Application Container Engine...
Oct 14 07:01:11 n3 systemd[1]: docker.service start request repeated too quickly, refusing to start.
Oct 14 07:01:11 n3 systemd[1]: Failed to start Docker Application Container Engine.
```

* Now the strange thing: Running `service docker start` after login starts the docker daemon without issue:

```
# service docker start
# journalctl -u docker
....
Oct 14 07:01:11 n3 systemd[1]: Starting Docker Application Container Engine...
Oct 14 07:01:11 n3 systemd[1]: docker.service start request repeated too quickly, refusing to start.
Oct 14 07:01:11 n3 systemd[1]: Failed to start Docker Application Container Engine.
Oct 14 07:01:59 n3 systemd[1]: Starting Docker Application Container Engine...
Oct 14 07:01:59 n3 dockerd[854]: time="2017-10-14T07:01:59.911404055Z" level=warning msg="[!] DON'T BIND ON ANY IP ADDRESS WITHOUT setting --tlsverify IF YOU DON'T KNOW WHAT YOU'RE DOING [!]"
Oct 14 07:01:59 n3 dockerd[854]: time="2017-10-14T07:01:59.924292299Z" level=info msg="libcontainerd: new containerd process, pid: 861"
Oct 14 07:02:01 n3 dockerd[854]: time="2017-10-14T07:02:01.001362136Z" level=info msg="[graphdriver] using prior storage driver: overlay2"
Oct 14 07:02:01 n3 dockerd[854]: time="2017-10-14T07:02:01.045499570Z" level=info msg="Graph migration to content-addressability took 0.00 seconds"
Oct 14 07:02:01 n3 dockerd[854]: time="2017-10-14T07:02:01.051642305Z" level=info msg="Loading containers: start."
Oct 14 07:02:02 n3 dockerd[854]: time="2017-10-14T07:02:02.007439431Z" level=info msg="Default bridge (docker0) is assigned with an IP address 172.17.0.0/16. Daemon option --bip can be used to set a preferred IP address"
Oct 14 07:02:02 n3 dockerd[854]: time="2017-10-14T07:02:02.291170107Z" level=info msg="Loading containers: done."
Oct 14 07:02:02 n3 dockerd[854]: time="2017-10-14T07:02:02.586186232Z" level=info msg="Docker daemon" commit=afdb6d4 graphdriver(s)=overlay2 version=17.09.0-ce
Oct 14 07:02:02 n3 dockerd[854]: time="2017-10-14T07:02:02.589064651Z" level=info msg="Daemon has completed initialization"
Oct 14 07:02:02 n3 dockerd[854]: time="2017-10-14T07:02:02.676636595Z" level=info msg="API listen on 192.168.23.203:2375"
Oct 14 07:02:02 n3 systemd[1]: Started Docker Application Container Engine.
Oct 14 07:02:02 n3 dockerd[854]: time="2017-10-14T07:02:02.683913493Z" level=info msg="API listen on /var/run/docker.sock"
```

* Removing the external port and using the following `daemon.json` works fine after reboot, though

```
{
  "hosts": [ "fd://" ]
}
```

------------------

Any idea why docker can't bind to the given address and port during system startup ? Any security related issue maybe ?

rhuss, 2017-10-14:
This is related to the issues we have in https://github.com/Project31/ansible-kubernetes-openshift-pi3/issues/28 but occurs for 17.09 as well as for 17.03

StefanScherer, 2017-10-14:
I've tried exactly your steps with 17.05.0-ce and 17.03.0-ce (HypriotOS 1.4.0) on another RPi3 and it worked for me. Rebooted my Pi and can use unix socket and TCP port.

rhuss, 2017-10-14:
I used Hypriot 1.6.0. Might this be the difference ? I flashed 1.6.0 and performed directly the steps above after flashing (also on RPi 3). It's completely reproducible for me ;-( 

Really wonder why this 

```
listen tcp 192.168.23.203:2375: bind: cannot assign requested address
```

happens during bootup (same when I change to 2376 using certs)

rhuss, 2017-10-14:
Actually tried it with Hypriot OS 1.4.0 and it worked ! So looks really like an issue with Hypriot OS 1.6.0. Going to try now 1.5.1, too ...

firecyberice, 2017-10-14:
maybe the network is not up at dockerd startup.
Another option is binding to ip 0.0.0.0:2375.

rhuss, 2017-10-14:
> Actually tried it with Hypriot OS 1.4.0 and it worked ! So looks really like an issue with Hypriot OS 1.6.0. Going to try now 1.5.1, too ...

1.5.1 fails, too, when binding to the concrete IP. 

@firecyberice that's it ! Binding to all interfaces works during startup, works perfectly !!

However, something seems to have be changesd between Hypriot 1.4.0 and 1.5.1 wrt to network setup which prevents binding to a concrete IP.

The current solution is good enough for me. Thanks a lot, guys ! 'hope to see you soon again :) ...

closed 2017-10-14 by rhuss

rscai, 2018-03-17:
I encountered same issue, resolved by edit `/lib/systemd/system/docker.service`.

`docker.service` hard code argument `-H fd://` for `/usr/bin/dockerd`, caused conflict with `/etc/docker/daemon.json`. On /lib/systemd/system/docker.service, found

```
[Service]
Type=notify
# the default is not to use systemd for cgroups because the delegate issues still
# exists and systemd currently does not support the cgroup feature set required
# for containers run by docker
ExecStart=/usr/bin/dockerd -H fd://
ExecReload=/bin/kill -s HUP $MAINPID
LimitNOFILE=1048576
```

Change `ExecStart=/usr/bin/dockerd -H fd://` to `ExecStart=/usr/bin/dockerd`. 
Reload `docker.service`

```shell
systemctl daemon-reload
```

Then restart docker daemon

```shell
service docker restart
```

**System and Software Information**

Name|Version
----|-------
OS  | Ubuntu 16.04.4 LTS (GNU/Linux 4.4.0-116-generic x86_64)
Docker| 17.12.1-ce, build 7390fc6