hypriot/image-builder-rpi #191: Wrong date and time after first boot

lukasmalkmus, 2017-10-25:
Right after flashing my first HypriotOS image and starting up my Pi, I noticed that date and time do not match. I noticed this after `curl`, `apt`, etc. complained that some SSL certificates were not valid yet.

This is the output of `date` after flashing a fresh image and adding my ssh key:
```bash
Thu Jul  3 18:50:40 UTC 2014
```

I tested it on two RPi3s.

lukasmalkmus, 2017-10-25:
Looks like `timesyncd` is having trouble:

```bash
systemctl status systemd-timesyncd
```
```bash
● systemd-timesyncd.service - Network Time Synchronization
   Loaded: loaded (/lib/systemd/system/systemd-timesyncd.service; enabled)
   Active: active (running) since Thu 2014-07-03 20:46:22 CEST; 13min ago
     Docs: man:systemd-timesyncd.service(8)
 Main PID: 271 (systemd-timesyn)
   Status: "Using Time Server 212.112.228.242:123 (3.debian.pool.ntp.org)."
   CGroup: /system.slice/systemd-timesyncd.service
           └─271 /lib/systemd/systemd-timesyncd

Jul 03 20:59:29 node-01 systemd-timesyncd[271]: Timed out waiting for reply from 195.50.171.101:123 (2.debian.pool.ntp.org).
Jul 03 20:59:29 node-01 systemd-timesyncd[271]: Using NTP server [2a01:4f8:172:1eee::2]:123 (2.debian.pool.ntp.org).
Jul 03 20:59:29 node-01 systemd-timesyncd[271]: Using NTP server [2001:638:504:2000::32]:123 (2.debian.pool.ntp.org).
Jul 03 20:59:29 node-01 systemd-timesyncd[271]: Using NTP server [2a01:4f8:13b:3ba6::2]:123 (2.debian.pool.ntp.org).
Jul 03 20:59:29 node-01 systemd-timesyncd[271]: Using NTP server [2001:1640:3:4::dead:beef]:123 (2.debian.pool.ntp.org).
Jul 03 20:59:29 node-01 systemd-timesyncd[271]: Using NTP server 78.46.204.247:123 (3.debian.pool.ntp.org).
Jul 03 20:59:40 node-01 systemd-timesyncd[271]: Timed out waiting for reply from 78.46.204.247:123 (3.debian.pool.ntp.org).
Jul 03 20:59:40 node-01 systemd-timesyncd[271]: Using NTP server 176.9.144.121:123 (3.debian.pool.ntp.org).
Jul 03 20:59:50 node-01 systemd-timesyncd[271]: Timed out waiting for reply from 176.9.144.121:123 (3.debian.pool.ntp.org).
Jul 03 20:59:50 node-01 systemd-timesyncd[271]: Using NTP server 212.112.228.242:123 (3.debian.pool.ntp.org).
Hint: Some lines were ellipsized, use -l to show in full.
```

But `ping 212.112.228.242` works as expected.

StefanScherer, 2018-03-22:
Firewall etc?
Did you solve your problem?

lukasmalkmus, 2018-04-09:
Sorry for not catching your response. No, nothing fancy like a firewall.
Last time I checked, it was not working. Will try again and check.

closed 2019-09-16 by StefanScherer