hypriot/image-builder-rpi #198: v1.7.1 has cloud-init version 0.7.6 installed instead of 0.7.9

philstel, 2017-11-09:
The release v1.7.1 of hypriot has version 0.7.6 of cloud-init installed and not 0.7.9 as described in
`image-builder-rpi/builder/files/boot/user-data` => `# The current version of cloud-init in the Hypriot rpi-64 is 0.7.9`

I suspect you took the file directly from
`  https://github.com/DieterReuter/image-builder-rpi64/releases/download/v20171013-172949/hypriotos-rpi64-v20171013-172949.img.zip` were indeed version 0.7.9 is installed.

Most of the stuff in the example user-data file does not work with version 0.7.6 and there is also no documentation for this version in the [official docs](http://cloudinit.readthedocs.io/en/latest). Oldest version i can find there is 0.7.6.

I can provide a pull request if you can hint me in the right direction to fix this. Is there a more recent version in any repository available we can use?

DieterReuter, 2017-11-09:
Thanks. You're totally right, will change that in the comments.

philstel, 2017-11-09:
So it's not possible to upgrade to 0.7.9? Version 0.7.6 is really limited to the point were its barely usable (at least for my use case).
Do you use other apt repositories for `https://github.com/DieterReuter/image-builder-rpi64` ?

DieterReuter, 2017-11-09:
I just use the standard APT repos. For 64bit it's based upon Debian, for 32bit it's based upon Raspbian (to support ARMv6 and RPi Zero). I guess that's the main difference. So you could check how it is possible to use a new version in Raspbian!

sensorii, 2017-11-20:
I looked at this issue and thought to mi self, what would a pirate do?...

Install the latest greatest version, of course!  which is not 0.7.6 or 0.7.9, it is more like cloud-init version 17.1.

So down the rabbit hole I went in search of cloud-init v17.1 for RPi and ended up building the debian package from source... hacking the way to success: [rpi-cloud-init-builder](https://github.com/sensorii/rpi-cloud-init-builder)

I install the .deb package in image-builder-rpi [like this](https://github.com/sensorii/image-builder-rpi/commit/7f81aea33f617d2a82e84de001c3001b245ca9cd).

It builds, passes all cloud-init tests, and runs.  cloud-init v17.1 processes my custom `/boot/user-data` (same as 0.7.9 syntax) on startup, and it works!

It can be improved... there is a warning on login (sensorii/rpi-cloud-init-builder#1) with an easy way to shush it, but worth investigating proper fix.

I don't have much experience building deb packages for distribution, not sure what to do with the files from here, there could be better ways.  So I present this in case another pirate wants to carry on with this as a possible solution to get the latest cloud-init versions into HypriotOS images.  It's easy to configure the builder to build other cloud-init versions, too.

Maybe related (mentions v17.1): DieterReuter/image-builder-rpi64#31

sensorii, 2017-11-21:
I'm finding more things to polish to integrate cloud-init 17.1.  I think my first success was random (where I could ssh in as the created user, which is now failing). ssh daemon keys go missing more times than not, as both cloud-init and hypriot-firstboot regen the keys and the later fails.  all fixable things, further down the rabbit hole... :rabbit:

StefanScherer, 2017-11-21:
I‘m not keen on adding a self compiled version of cloud-init. One of the reasons we switched from device-init to cloud-init is that we don‘t have to maintain it. So we choose the available versions out there.

StefanScherer, 2017-12-18:
Comment fixed with 5f3e8d2cdd26b7347e18b36bcfe5f5d6a2765a58

closed 2017-12-18 by StefanScherer