hypriot/image-builder-rpi #201: setting multiple wlan connections in device-init.yaml

sdorosz, 2017-11-20:
Is there a way already to define mutiple ssid/password pairs for wlan connections in the device-init.yaml file?

firecyberice, 2017-11-29:
We replaced device-init with cloud-init in release 1.7.1 because it has more config options.
For more information about device-init please have a look at [device-init](https://github.com/hypriot/device-init).

StefanScherer, 2017-12-18:
Closing this in favor of cloud-init. Please have a look at the example user-data file for the cloud-init first boot: https://github.com/hypriot/image-builder-rpi/blob/5f3e8d2cdd26b7347e18b36bcfe5f5d6a2765a58/builder/files/boot/user-data#L45-L71
It should be possible to configure a second WiFi connection there as well.

closed 2017-12-18 by StefanScherer