hypriot/image-builder-rpi #202: Where is device-init on v1.7.1?

potatop, 2017-11-20:
There is no device-init at /usr/local/bin, and device-init.yaml file at /boot has no effect.

StefanScherer, 2017-11-21:
Exactly. Cloud-init is more flexible and better known and there are more docs out there. 
Have a look at the latest blog post and the /boot/user-data file.

potatop, 2017-11-22:
Was this stated some where? It's not clear from the release notes.

StefanScherer, 2017-11-22:
Oh you‘re right, we should mention it in https://github.com/hypriot/image-builder-rpi/releases/tag/v1.7.1 intoduced with 1.7.0 and some housekeeping in the download page. 
I‘ll reopen this issue for these tasks.

StefanScherer, 2017-11-29:
I've added 

```
## Breaking change
- No more device-init, please convert your `/boot/device-init.yaml` to `/boot/user-data`
```

to the 1.7.0 and 1.7.1 releases.

closed 2017-11-29 by StefanScherer

maxnowack, 2018-01-08:
There also should be a mention about that change in the readme of this repository: https://github.com/hypriot/device-init