hypriot/image-builder-rpi #203: Failed make sd-image

fudekun, 2017-12-06:
I am generating a unique Hypriot image.
For my change, I added multiple apt-get commands to `image-builder-rpi/builder/chroot-script.sh`.
I ran the `make sd-image`. However, I will fail on the way. The messages that came out at that time are as follows.
Please tell me how to avoid it.

```sh
:
:
+ sha256sum -c -
+ echo '2fbeb13b7b0f2308dbd0d82780b54c33003ad43d145ff08498b25fb8bbe1c2c6 /workspace/rpi-raw.img.zip'
/workspace/rpi-raw.img.zip: OK
+ unzip -p /workspace/rpi-raw.img
+ guestfish -a /hypriotos-rpi-dirty.img
^[[1;2C*stdin*:4: libguestfs: error: tar_in: tar subcommand failed on directory: /: tar: ./lib/arm-linux-gnueabihf/libparted.so.2.0.1: Wrote only 9728 of 10240 bytes
tar: ./lib/arm-linux-gnueabihf/libacl.so.1.1.0: Cannot write: No space left on device
tar: ./lib/arm-linux-gnueabihf/libaio.so.1.0.1: Cannot write: No space left on device
tar: ./lib/arm-linux-gnueabihf/libc-2.19.so: Cannot write: No space left on device
tar: ./lib/arm-linux-gnueabihf/libsystemd.so.0.3.1: Cannot write: No space left on device
tar: ./lib/arm-linux-gnueabihf/libply.so.2.1.0: Cannot write: No space left on device
tar: ./lib/arm-linux-gnueabihf/libgpg-error.so.0.13.0: Cannot write: No space left on device
tar: ./lib/arm-linux-gnueabihf/libdevmapper.so.1.02.1: Cannot write: No space left on device
tar: ./lib/arm-linux-gnueabihf/libhistory.so.6.3: Cannot write: No space left on device
tar: ./lib/arm-linux-gnueabihf/libkmod.so.2.2.8: Cannot write: No space left on device
:
:
```

fudekun, 2017-12-06:
Additional comments:
If I have not made any changes to `image-builder-rpi / builder / chroot-script.sh`, I will succeed in the build.

firecyberice, 2017-12-06:
If you would like to install additional packages I recomment to use cloud-init then it gets installed at boot time. Otherwise you have to expand the SD card partitioning layout.

fudekun, 2017-12-06:
I appreciate your answer.

closed 2017-12-06 by fudekun