hypriot/image-builder-rpi #204: Docker stuck on 17.05.0-ce (2017-05-04)

NargiT, 2017-12-12:
Hello,

My version of docker is stuck on 17.05.0-ce version but the latest is 17.11 and even the stable version is 17.09 (source : https://docs.docker.com/release-notes/docker-ce/)

I don't understand why 
`sudo apt-get update && sudo apt-get upgrade` do not upgrade :( I also tried dist-upgrade but nothing...

any idea ?

StefanScherer, 2017-12-13:
Docker has changed the repo and package name. I don‘t know a safe upgrade path. 

To install current versions you can use

```
curl https://get.docker.com | sh
```

But I don‘t know if you loose your images.

NargiT, 2017-12-15:
Yes you were right

I updated /etc/apt/sources.list.d/docker.list by 
deb [arch=armhf] https://download.docker.com/linux/raspbian jessie stable
and I have now the version I wanted
thx

closed 2017-12-15 by NargiT