hypriot/image-builder-rpi #211: Does the image use SELinux or AppArmor?

scud3r1a, 2018-01-01:
Does the image use SELinux or AppArmor? If not, should it?

StefanScherer, 2018-01-01:
I'm not aware that we have added SELinux or AppArmor. What is your use-case?
Yes, the SD card image is based on Raspbian, the rootfs is built at https://github.com/hypriot/os-rootfs
Some tools are missing and due to the other kernel you may find issues with additional hardware.

scud3r1a, 2018-01-01:
Thank you very much for your quick answer. I cannot name a specific use-case. But I read about the discussion to connect e.g. AppArmor and Docker multiple times. Would it make the system safer in your opinion?

StefanScherer, 2018-01-01:
Yes, the docker-ce packages installs libapparmor as dependency https://travis-ci.org/hypriot/image-builder-rpi/jobs/322848909#L2123

But it is up to the user to use the Docker options to run containers more securely.

scud3r1a, 2018-01-01:
So you have not set that up yet in general? For other applications?
Can you guess if Raspbian uses AppArmor by default?

StefanScherer, 2018-03-22:
No I don't know. Please use https://gitter.im/hypriot/talk for more questions / discussions. :-)

closed 2018-03-22 by StefanScherer