hypriot/image-builder-rpi #213: Unable to configure wlan

vikas027, 2018-02-05:
### Environment
Raspberry Pi: 3 Model B
OS: [hypriotos-rpi-v1.7.1.img.zip](https://github.com/hypriot/image-builder-rpi/releases/download/v1.7.1/hypriotos-rpi-v1.7.1.img.zip)
Flash Command: `flash --hostname node01 /path/to/hypriotos-rpi-v1.7.1.img.zip`

### Problem
Unable to configure wlan

I was able to configure WLAN easily in Raspbian OS using `/etc/dhcpcd.conf` as shown [here](https://raspberrypi.stackexchange.com/questions/37920/how-do-i-set-up-networking-wifi-static-ip-address) on the same Raspberry Pi.

#### Wireless Interface Details
```
$ iwconfig wlan0
wlan0     IEEE 802.11bgn  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=31 dBm
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on
$
```

Scanning SSIDs
```
$ iwlist wlan0 scan | head -n 15
wlan0     Scan completed :
          Cell 01 - Address: 20:4E:7F:20:00:C9
                    Channel:6
                    Frequency:2.437 GHz (Channel 6)
                    Quality=51/70  Signal level=-59 dBm
                    Encryption key:on
                    ESSID:"MYSSID-027"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 18 Mb/s
                              24 Mb/s; 36 Mb/s; 54 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 48 Mb/s
                    Mode:Master
                    Extra:tsf=0000000000000000
                    Extra: Last beacon: 70ms ago
                    IE: Unknown: 00084A4B564B2D322E34
                    IE: Unknown: 010882848B962430486C
$
```

### Configuration Files
```
$ vi /etc/wpa_supplicant/wpa_supplicant.conf
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
	ssid="MYSSID-027"
	psk=password_hash
	proto=RSN
	key_mgmt=WPA-PSK
	pairwise=CCMP
	auth_alg=OPEN}
$
$ chmod 0600 /etc/wpa_supplicant/wpa_supplicant.conf

$ vi /etc/network/interfaces.d/wlan0
allow-hotplug wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp
$
```

### Logs
There are no errors on console or `/var/log/syslog` but IP doesn't get configured
```
$ ifdown wlan0
ifdown: interface wlan0 not configured
$ ifup wlan0
$ ifconfig wlan0
wlan0     Link encap:Ethernet  HWaddr b8:27:eb:91:ef:57
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:120 errors:0 dropped:120 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:46872 (45.7 KiB)  TX bytes:0 (0.0 B)
$

$ tail -f /var/log/syslog
Feb  5 12:51:08 nextcloud wpa_supplicant[1088]: Successfully initialized wpa_supplicant
```

firecyberice, 2018-02-05:
Add your dhcp config to `/etc/network/interfaces.d/wlan0` and try again.
Feel free to reopen this issue again if it is not working.

closed 2018-02-05 by firecyberice

vikas027, 2018-02-05:
@firecyberice Can you please give an example and update the [user data](https://github.com/hypriot/image-builder-rpi/blob/5f3e8d2cdd26b7347e18b36bcfe5f5d6a2765a58/builder/files/boot/user-data#L45-L71) here as well?

firecyberice, 2018-02-06:
Please have a look at: https://github.com/hypriot/blog/issues/60

StefanScherer, 2018-03-22:
The next HypriotOS 1.8.0 will be easier to activate WiFi.

orubel, 2018-03-22:
I thought that adding this to your YAML files was supposed to take care of this? I have been trying a few dozen different things and this isn't working for me either. Have most recent version of flash.

By the way, great work. Loving everything so far. Issues are very few.

StefanScherer, 2018-03-23:
With HypriotOS up to 1.7.1 you have to change the /boot/config.txt and turn off uart. Hypriot 1.8.0 will have another default so you don't have to care about that.