hypriot/image-builder-rpi #215: linux-headers includes x86_64 tools...

brendandburns, 2018-02-18:
Trying to compile a kernel module with DKMS fails because tools like `scripts/recordmcount` are compiled for x86_64 not ARM...

This is due to cross-compiling on x86.  We should probably build the package on rpi so that these tools are also rpi compatible.

For now you can re-compile them individually as a workaround...

StefanScherer, 2018-02-19:
Thanks @brendandburns for "flying" HypriotOS :-)

I tried to reproduce the problem by installing the kernel headers we also provide as a package.
I flashed the latest HypriotOS 1.7.1 and did the following steps:

```
$ sudo apt-get update
$ sudo apt-get install raspberrypi-kernel-headers
$ /usr/src/linux-headers-4.4.50-v7+/scripts/recordmcount
usage: recordmcount [-w] file.o...
```

The tool shows me the usage, so I think this is compiled for ARM. Maybe you have installed the linux headers from a different package?

Our RPI kernel is built in repo https://github.com/hypriot/rpi-kernel and the deb packages are deployed to https://packagecloud.io/Hypriot/rpi

So if a kernel module is missing, maybe we should add this to our pipeline?

brendandburns, 2018-02-20:
I will try this again this evening...  I'm trying to build the iscsitarget module... Not sure if we want to include that or not...

--brendan

brendandburns, 2018-02-25:
To use DKMS, I believe I need the following installed:

linux-headers-4.4.50-hypriotos-v7+_4.4.50-hypriotos-v7+-1_armhf.deb

https://packagecloud.io/Hypriot/rpi/packages/debian/jessie/linux-headers-4.4.50-hypriotos-v7+_4.4.50-hypriotos-v7+-1_armhf.deb

which has the x86_64 tools inside it...

Many thanks for any help here...

brendandburns, 2018-03-10:
@StefanScherer 

friendly ping on this if there is any way for me to fix/recompile this package for arm...

Thanks!
--brendan

StefanScherer, 2018-03-10:
Thanks @brendandburns for the ping. I was down with influenza short after my last reply and also had to cancel my trip to MVP Summit this week. Otherwise we had the chance to hack on this in Redmond ;-)

What I don't understand is, I can't see any x86_64 tools inside this package. As mentioned above I've installed the package (with `sudo apt-get update && sudo apt-get install raspberrypi-kernel-headers`) on a Raspberry Pi 3 with HypriotOS 1.7.1 installed.
When I check all tools that are in the scripts folders

```
$ cd /usr/src/linux-headers-4.4.50+/scripts
$ file * | grep ELF
conmakehash:                       ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=2c3a1e539429dd1d4b238562decb326faeb06e2c, not stripped
kallsyms:                          ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=94af920e5874f37a9d745ca4ba063dd06dc5b832, not stripped
pnmtologo:                         ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=31b9263651369b4f570f3a1209b9c480ec4ee424, not stripped
recordmcount:                      ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=2a87ecdd01da5e542f3e55d679e401c3d9a2a5f9, not stripped
sortextable:                       ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=b28c91f1371f766790d2f6cd69b3ad4ea2f83a3f, not stripped
HypriotOS/armv7: pirate@black-pearl in /usr/src/linux-headers-4.4.50+/scripts
$ cd ../..
HypriotOS/armv7: pirate@black-pearl in /usr/src
$ cd linux-headers-4.4.50-v7+/scripts/
HypriotOS/armv7: pirate@black-pearl in /usr/src/linux-headers-4.4.50-v7+/scripts
$ file * | grep ELF
conmakehash:                       ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=2c3a1e539429dd1d4b238562decb326faeb06e2c, not stripped
kallsyms:                          ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=94af920e5874f37a9d745ca4ba063dd06dc5b832, not stripped
pnmtologo:                         ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=31b9263651369b4f570f3a1209b9c480ec4ee424, not stripped
recordmcount:                      ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=2a87ecdd01da5e542f3e55d679e401c3d9a2a5f9, not stripped
sortextable:                       ELF 32-bit LSB executable, ARM, EABI5 version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-armhf.so.3, for GNU/Linux 2.6.32, BuildID[sha1]=b28c91f1371f766790d2f6cd69b3ad4ea2f83a3f, not stripped
```

every ELF binary is compiled for ARM.

Do you have any command I should check in my environment to see the error you have?

To compile this package I think the most lazy way would be to fork https://github.com/hypriot/rpi-kernel and activate the CircleCI build in that fork. Then you can adjust the two files
* [kernel_configs/rpi1_docker_kernel_config](https://github.com/hypriot/rpi-kernel/blob/master/kernel_configs/rpi1_docker_kernel_config)
* [kernel_configs/rpi2_3_docker_kernel_config](https://github.com/hypriot/rpi-kernel/blob/master/kernel_configs/rpi2_3_docker_kernel_config)
in the fork and let Circle build the Linux kernel and kernel modules and build the deb packages.

Circle is configured to store all deb files as artifacts, see our last [build 59](https://circleci.com/gh/hypriot/rpi-kernel/59) in master branch and the corresponding artifacts:

[image]

So you are able to download the deb packages to your Pi and let a cloud server do the dirty task of compiling.

The compilation is done in the [scripts/compile_kernel.sh](https://github.com/hypriot/rpi-kernel/blob/master/scripts/compile_kernel.sh), of course cross compiled on an x86_64 machine on Circle. 

I haven't compiled the kernel natively on a Raspberry Pi for years, the Raspberry Pi docs have a page [Kernel Building](https://www.raspberrypi.org/documentation/linux/kernel/building.md).

brendandburns, 2018-03-19:
Ok, with the latest headers I don't see this.  I don't know what the issue was...

But sadly I'm not building compatible modules still, I will send a repro in a different issue.

closed 2018-03-19 by brendandburns