hypriot/image-builder-rpi #216: Remove upstream packages

miekg, 2018-02-21:
Hi,

[not sure if this is the right repo to file the issue]

As hypriot is only(?) used to run docker containers and or kubernetes, I see I bunch of installed packages that are superfluous to this purpose. I.e. running (today) an `apt-get -u update && apt-get -u upgrade`:
~~~
$ sudo apt-get -u upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  docker-ce
The following packages will be upgraded:
  bind9-host curl gcc-4.9-base kubeadm kubectl kubelet libbind9-90 libcurl3 libcurl3-gnutls libdns-export100 libdns100 libgcc1 libirs-export91 libisc-export95 libisc95
  libisccc90 libisccfg-export90 libisccfg90 liblwres90 libstdc++6 libxml2
21 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
~~~
Note the primary reason I care is that updating needs to write to my poor SD cards. Does it make sense to trim these images?

StefanScherer, 2018-03-22:
I've added `--no-install-recommends` to the installation of the `docker-ce` package with https://github.com/hypriot/image-builder-rpi/pull/220/files#diff-7c1dca149a28360b3302bc525fd991a3R232 which keeps some other packages off the SD image.

StefanScherer, 2018-03-22:
If you like you can try https://github.com/hypriot/image-builder-rpi/releases/tag/v1.8.0-rc1 and give use feedback :-)

closed 2019-09-16 by StefanScherer