hypriot/image-builder-rpi #217: cloud-init 0.7.6 instead of 0.7.9 since november

Bootsektor, 2018-02-26:
and no fix or no mention on the blog page

tIsGoud, 2018-03-12:
Latest documentation on cloud-init is now version 0.7.7, that is another reason to update it.

StefanScherer, 2018-03-22:
What's the exact problem here? Just the version numbers?

tIsGoud, 2018-03-24:
@StefanScherer the cloud-config (user-data) file refers to 0.79 for the documentation. 
From what @bootsektor writes I understand that the November version had cloud-init 0.79 installed.

The actual version of cloud-init on the hypriot image (hypriotos-rpi-v1.7.1.img) is 0.76.

StefanScherer, 2018-03-25:
Just flash a 1.7.1 image. Yes, cloud-init package is 0.7.6 due to Raspbian Jessie. We're preparing a Raspbian Stretch release with HypriotOS 1.8.0, there will be cloud-init 0.7.9 installed.

closed 2018-03-25 by StefanScherer