hypriot/image-builder-rpi #219: Building an incompatible module?

brendandburns, 2018-03-19:
Here are the repro steps:

`apt-get install iscsitarget-dkms`

The build will break, because the Debian package is too old.

So:

```
cd /usr/src/iscsitarget-1.4.20.3+svn502
curl https://gist.githubusercontent.com/pcarranza/df74e8aea88436298692828eb74501ba/raw/78fde3aa00ae696d7c9bdd16621a23865e36370c/iscsitarget-dkms_1.4.20.3+svn502-1.patch > work.patch
patch -p1 < work.patch

dkms build iscsitarget/1.4.20.3+svn502
dkms install iscsitarget/1.4.20.3+svn502

modprobe iscsi_trgt
```

This errors out and if you run `dmesg` you see:

```
iscsi_trgt disagrees about version of symbol module_layout
```

Given that the headers are right and the module built, I would have thought this should work...

Any suggestions welcome.

Thanks!

StefanScherer, 2018-03-25:
Tried the steps on HypriotOS v1.7.1 and now got the same error message. 
```
modprobe iscsi_trgt
modprobe: ERROR: could not insert 'iscsi_trgt': Exec format error
```

and `dmesg` with message `iscsi_trgt: disagrees about version of symbol module_layout`

The kernel module itself is compiled for ARM.

Maybe the answer here is a hint https://unix.stackexchange.com/questions/125624/unable-to-load-module-disagrees-about-version-of-symbol-module-layout 
Sounds like you have to compile the kernel to have all files needed for the same exact data structure.

StefanScherer, 2020-03-14:
Closing due to inactivity.

closed 2020-03-14 by StefanScherer