hypriot/image-builder-rpi #223: 1.8.0-rc1: multiple IP addresses for eth0, docker swarm init complains

StefanScherer, 2018-03-23:
Test with HypriotOS 1.8.0-rc1.

$ ssh pirate@ethtst.local
Warning: the ECDSA host key for 'ethtst.local' differs from the key for the IP address 'fe80::ba27:ebff:feb9:81de%en0'
Offending key for IP in /Users/stefan/.ssh/known_hosts:43
Matching host key in /Users/stefan/.ssh/known_hosts:296
Are you sure you want to continue connecting (yes/no)? yes
pirate@ethtst.local's password: 
Linux ethtst 4.9.80-hypriotos-v7+ #1 SMP Wed Mar 21 22:53:39 UTC 2018 armv7l

HypriotOS (Debian GNU/Linux 9)

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Thu Mar 22 09:08:45 2018 from fe80::107c:3301:8c32:f2ec%eth0
HypriotOS/armv7: pirate@ethtst in ~
$ ifconfig
docker0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 172.17.0.1  netmask 255.255.0.0  broadcast 172.17.255.255
        ether 02:42:c3:76:d0:58  txqueuelen 0  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.221  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::ba27:ebff:feb9:81de  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:b9:81:de  txqueuelen 1000  (Ethernet)
        RX packets 63  bytes 8988 (8.7 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 91  bytes 14660 (14.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 72  bytes 5766 (5.6 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 72  bytes 5766 (5.6 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:ec:d4:8b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

HypriotOS/armv7: pirate@ethtst in ~
$ docker swarm init
Error response from daemon: could not choose an IP address to advertise since this system has multiple addresses on interface eth0 (192.168.1.221 and 192.168.1.213) - specify one with --advertise-addr
HypriotOS/armv7: pirate@ethtst in ~
$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:b9:81:de brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.221/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet 192.168.1.213/24 brd 192.168.1.255 scope global secondary eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:feb9:81de/64 scope link 
       valid_lft forever preferred_lft forever
3: wlan0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc pfifo_fast state DOWN group default qlen 1000
    link/ether b8:27:eb:ec:d4:8b brd ff:ff:ff:ff:ff:ff
4: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 02:42:c3:76:d0:58 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
HypriotOS/armv7: pirate@ethtst in ~
$ ls -l /etc/network/interfaces.d
total 8
-rw-r--r-- 1 root root 367 Mar 22 09:08 50-cloud-init.cfg
-rw-r--r-- 1 root root  40 Mar 20 08:09 eth0
HypriotOS/armv7: pirate@ethtst in ~
$ cat /etc/network/interfaces.d/50-cloud-init.cfg 
# This file is generated from information provided by
# the datasource.  Changes to it will not persist across an instance.
# To disable cloud-init's network configuration capabilities, write a file
# /etc/cloud/cloud.cfg.d/99-disable-network-config.cfg with the following:
# network: {config: disabled}
auto lo
iface lo inet loopback

auto eth0
iface eth0 inet dhcp
HypriotOS/armv7: pirate@ethtst in ~
$ cat /etc/network/interfaces.d/eth0 
allow-hotplug eth0
iface eth0 inet dhcp
HypriotOS/armv7: pirate@ethtst in ~

StefanScherer, 2018-03-23:
Trying to fix that with #224 (artifact with SD card image available at [circle build 37](https://circleci.com/gh/hypriot/image-builder-rpi/37?utm_campaign=vcs-integration-link&utm_medium=referral&utm_source=github-build-link))

FYI @orubel

StefanScherer, 2018-03-23:
Hm, even with this disable I get two IP addresses

```
HypriotOS/armv7: pirate@black-pearl in ~
$ ls /etc/network/interfaces.d
eth0
HypriotOS/armv7: pirate@black-pearl in ~
$ ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether b8:27:eb:2f:a1:b8 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.221/24 brd 192.168.1.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet 192.168.1.212/24 brd 192.168.1.255 scope global secondary eth0
       valid_lft forever preferred_lft forever
    inet6 fe80::ba27:ebff:fe2f:a1b8/64 scope link 
       valid_lft forever preferred_lft forever
3: docker0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default 
    link/ether 02:42:7a:51:86:f9 brd ff:ff:ff:ff:ff:ff
    inet 172.17.0.1/16 brd 172.17.255.255 scope global docker0
       valid_lft forever preferred_lft forever
$ docker swarm init
Error response from daemon: could not choose an IP address to advertise since this system has multiple addresses on interface eth0 (192.168.1.221 and 192.168.1.212) - specify one with --advertise-addr
```

zerodivide1, 2018-03-24:
Hi,
I was having some difficulty setting a static IP address with 1.8.0-rc1 and found this to work for me:

1. Write `/etc/network/interfaces.d/eth0` with the following:
  ```
  auto eth0
  #allow_hotplug eth0
  iface eth0 inet static
  address 192.168.1.18
  netmask 255.255.255.0
  network 192.168.1.0
  broadcast 192.168.1.255
  gateway 192.168.1.1
  ```
2. Add the `/etc/cloud/cloud.cfg/99-disable-network-config.cfg` with the following:
  ```
  network: {config: disabled}
  ```
3. Manually removed the `/etc/network/interfaces.d/50-cloud-init.cfg` file. 
4. Reboot

I performed the above on 2 brand new Pi 3 B+ and for each it seemed to work fine. Not sure how to perform the file removal above in the `user-data` cloud-init config file to set it up during first boot, however.

PR #224 (2018-03-25): Fix duplicate IP address for eth0
~Disable cloud-init network config, the /etc/network/interfaces.d/eth0 is already there.~
Update: The problem also exists if clout-init's network config is disabled.

We have to remove the /etc/network/interfaces.d/eth0 to fix the duplicate IP address for eth0.

Fixes #223
closed 2018-03-25 by StefanScherer