hypriot/image-builder-rpi #233: Failing service wifi-country.service on boot

Duvel, 2018-03-28:
In the newest raspbian they have a system in place to disable the wifi until the country is set. I guess this is not working in Hypriot as I see this service failing during startup:

```
$ sudo systemctl status wifi-country.service 
● wifi-country.service - Disable WiFi if country not set
   Loaded: loaded (/lib/systemd/system/wifi-country.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Wed 2018-03-28 17:19:36 CEST; 2min 13s ago
  Process: 320 ExecStart=/usr/lib/raspberrypi-sys-mods/wifi-country (code=exited, status=1/FAILURE)
 Main PID: 320 (code=exited, status=1/FAILURE)
      CPU: 27ms

Mar 28 17:19:36 walvis systemd[1]: Starting Disable WiFi if country not set...
Mar 28 17:19:36 walvis wifi-country[320]: rfkill not installed
Mar 28 17:19:36 walvis systemd[1]: wifi-country.service: Main process exited, code=exited, status=1/FAILURE
Mar 28 17:19:36 walvis systemd[1]: Failed to start Disable WiFi if country not set.
Mar 28 17:19:36 walvis systemd[1]: wifi-country.service: Unit entered failed state.
Mar 28 17:19:36 walvis systemd[1]: wifi-country.service: Failed with result 'exit-code'.
```

StefanScherer, 2018-03-29:
Oh, the package `rfkill` is missing. I have seen this service running during some first builds for 1.8.0.

Duvel, 2018-03-29:
They introduced this service in the latest version, I just disabled this service, because I don't like it blocking wifi without any notification when you're working headless. It did cost me sometime to get wifi up and I don't want that to happen again.

StefanScherer, 2018-03-30:
Tried to add rfkill, but then WiFi does not come up. I think I keep it as it is for now as I don't have the time to investigate this further.

closed 2019-09-16 by StefanScherer