hypriot/image-builder-rpi #241: How to enable ssh on Pi Zero's USB OTG?

ahirner, 2018-04-22:
I'm flashing 1.8.1 and love the workflow so far. However, I've found no way to use the OTG port for ssh access. There are three modifications necessary according to this [guide](https://gist.github.com/gbaman/975e2db164b3ca2b51ae11e45e8fd40a). Some of them cannot be persisted with cloud-init and the last one doesn't suffice either.

**1) Appending `dtoverlay=dwc2`to `config.txt`**
I'm flashing my own version with the `-c` option which yields the right file on `/boot`. Still there is `device-init.yaml` with vanilla content too. This might be unrelated but confusing. Also `gpu_mem` is reset to 16 no matter the value in the flashed file.

**2) Inserting `modules-load=dwc2,g_ether` into `cmdline.txt`**
I'm doing that manually after flashing. Is it correct that there is no way to flash this file? Would `runcmd` in user-data be the proper way for inserting that option permanently?

**3) Creating `/boot/ssh`**
After step 2, `usb0` shows up as an ethernet adapter and can be pinged. Presence of this file should enable ssh access. However, the file disappears after rebooting. Creating it after boot, yields access denied on the ssh port.
What is the right way to create additional files on `/boot/`? I've tried `write_files` with `content: ` or `content: abc` but it errs on boot (util.py[Warning]: "Failed loading yaml blob").
Is there an alternative approach to enable ssh over `usb0`?

PS: ssh over wifi works as intended.

StefanScherer, 2018-04-22:
I never tried OTG, but here are some hints that may help you.

1) you need the option `-C` (upper case). The `-c` option is for the old device-init.yaml file.

2) Oh, indeed we don't have a flash option to write the cmdline.txt file. Maybe we should add one in the https://github.com/hypriot/flash repo.

3) I doubt that we really need `/boot/ssh`. This is a feature of Raspbian to enable SSH. With cloud-init we automatically enable SSH without this file. Maybe there is missing some other package installed or a different configuration. Maybe check the packages installed on Raspbian and Hypriot, also check SSH configuration files between Raspbian and Hypriot.

So probably we find out what's missing and add it to Hypriot.

ahirner, 2018-04-24:
The capital `-C` option solved the first point. 
Before there is an official way, I updated cmdline.txt on first boot. For ssh to appear on the same address,  you need to write out a network interface file with cloud-init 0.79. After a final reboot, the Zero offers ssh via usb0 on the link-local address `169.254.1.1`. 
Here's the full user-data snippet:
```yaml
write_files:
   #USB OTG fixed link-local address
  - content: |
      allow-hotplug usb0
      iface usb0 inet static
          address 169.254.1.1
          netmask 255.255.0.0
    path: /etc/network/interfaces.d/usb0
runcmd:
  # Insert modules to load for usb0
  - sed 's/\<rootwait\>/& modules-load=dwc2,g_ether/' /boot/cmdline.txt -i
```
Issue solved, thanks!

StefanScherer, 2018-04-24:
Awesome. Thanks for sharing your solution.

shreyasnaik, 2018-06-04:
ahirner were you trying to get OTG working with Hypriot to get a k8s cluster working?

StefanScherer, 2019-09-16:
Closing due to inactivity

closed 2019-09-16 by StefanScherer