hypriot/image-builder-rpi #244: Hypriot 1.9.0 : how to preload docker images without device-init

samouds, 2018-05-07:
Hello,

I am wondering if it's possible to preload my docker image now that device-init is removed ?

I need to preload my app image for devices that can't have any web connection.

firecyberice, 2018-05-07:
Maybe you can use cloud-init instead. See [Breaking Changes on v1.9.0](https://github.com/hypriot/image-builder-rpi/releases)
- https://cloud-init.io/
- https://cloudinit.readthedocs.io/en/latest/

sensorii, 2018-05-20:
I'm also seeking a solution for this.  The key aspect is for this to work on first boot for devices **that can't have any web connection**.  My context is a headless IoT product that boots up a fully running application (distributed thru several docker images); ideally boots fast, has minimal SD card image for distribution, and does not require Internet connectivity.

It is true that cloud-init can load docker images, with an example `user-data` file in [this blog post](https://blog.hypriot.com/post/cloud-init-cloud-on-hypriot-x64/).  However, I think that requires a web connection... since the docker images are not preloaded on the SD card image.

I've tried including my docker image files (produced with `docker save`) in an SD card image built with a customized `image-builder-rpi`.  The idea is for the docker image files to be loaded (using `docker load`) from the SD card image upon first boot using cloud-init configuration. The image fails to build, it dies with tons of `No space left on device` errors:
```
guestfsd: error: tar subcommand failed on directory: /: tar: ./src/mydockerimage-rpi.tar: Wrote only 4608 of 10240 bytes
tar: ./src/tag: Cannot write: No space left on device
tar: ./proc: Cannot mkdir: No space left on device
tar: ./dev: Cannot mkdir: No space left on device
...
tar: Exiting with failure status due to previous errors
libguestfs: closing guestfs handle 0x560148605360 (state 2)
...
Makefile:7: recipe for target 'sd-image' failed
make: *** [sd-image] Error 1
```
Seems a size limit has been exceeded.  Maybe there is a related guestfish configuration option that could be increased(?), not something I understand yet.

I think the docker save/load approach isn't a bad idea, should work, and would work without web connection.  I don't consider it an ultimate solution, because `docker load` can be **sloooow** which translates to a slow first boot process, and I think it may be wasteful on space (since the saved images are an intermediary step to end up inside `/var/lib/docker`).  A more ideal solution would be a faster first boot process, with smaller SD card image, resulting in pre-loaded running docker images.

There's this idea of copying docker image files directly from `/var/lib/docker/aufs` and `/var/lib/docker/image` from a running machine that has already loaded the docker images for a more direct export/import, saving disk space and skipping load time:
* https://stackoverflow.com/questions/41154305/how-to-copy-docker-images-from-one-host-to-another/41276481#41276481
* https://askubuntu.com/questions/858536/how-to-load-docker-image-from-a-kickstart-or-preseed-file/863365#863365

I haven't tried that yet.  Would be great to automate an SD card image build process using that technique, it needs a running docker daemon somewhere to load the images first before copying them out to be included into SD card image.

Any other ideas or experience with solutions to accomplish this?

sensorii, 2018-05-25:
I worked thru the size limit by compressing my docker images further and `apt-get purge`'ing some larger packages that I didn't need installed on the SD card image.  I plan to explore [docker-slim](http://dockersl.im/) to decrease the docker image sizes even further. (already using [multi-stage docker builds](https://docs.docker.com/develop/develop-images/multistage-build/) and alpine images for smaller image size.)

Importing docker image files upon first boot is working OK.  The Docker export/import process is missing benefit of shared base images (since the entire image, including base(s) is exported) -- I haven't explored improving that yet.

I'm currently distributing docker image files using [zsync](http://zsync.moria.org.uk/) to minimize file transfer for device upgrades, with exploration of IPFS and [balena](https://www.balena.io/) on the roadmap.

Here's an example idea of pre-loading docker images onto HypriotOS SD card image, using docker export/import.  Works for devices without any web connection to boot into running docker images.

#### Preload and auto-start Docker images onto SD card image:

From a host that has the Docker images:
```bash
docker save myimage:mytag | gzip --best --rsyncable > myimage-mytag.tar.gz
```

Add the image to the SD card image and create a little script for loading the docker image(s).
Edit `image-builder-rpi/builder/chroot-script.sh`:
```bash
# import docker images
# (lots of different ways to do this, using wget if they are hosted somewhere)
mkdir /src && cd /src
wget https://example.com/myimage-mytag.tar.gz
wget https://example.com/docker-compose.yml

# create startup script
cat << EOF > /src/start.sh
#!/bin/bash
cd /src
docker load --input myimage-mytag.tar.gz
docker-compose up -d
EOF
chmod 770 /src/start.sh
```

Flash a cloud-init `user-data` file with a run command for loading the docker image(s):
```yml
# These commands will be ran once on first boot only
runcmd:
  # Pickup the hostname changes
  - 'systemctl restart avahi-daemon'
  # load and start docker image(s)
  - '/src/start.sh'
```

Tada!  Device will load and start running the Docker image(s) upon boot, no web connection required.

firecyberice, 2018-05-28:
Th change the image size please have a look [here](https://github.com/hypriot/image-builder-raw/blob/master/builder/rpi/build.sh#L12) The releases from this repo are getting combined with the [root filesystem](https://github.com/hypriot/os-rootfs) to provide an image ready to flash to SD cards.
So please build a raw image of your required size and change the references [here](https://github.com/hypriot/image-builder-rpi/blob/master/versions.config#L5-L8)

breandan, 2018-06-16:
Forgive my ignorance about how boot images work, but what prevents you from just copying the Docker image to `/var/lib/docker` after flashing the SD card? Do you need to update a checksum somewhere? Is there a way to make this process any simpler? @StefanScherer

DieterReuter, 2018-06-16:
The only way is to store the Docker image as a tar.gz file and import/install it with a ‘docker load’ command when the Pi is booted.

DieterReuter, 2018-06-16:
but you could do that on a Pi and then backup the section in /var/lib/docker/images or so and copy this over to a new image - this way it should be possible to pre-load a set of Docker images.

samouds, 2018-07-16:
@sensorii did you try to use  docker-slim ? or any other solution to decrease the image size ?

samouds, 2018-07-19:
@firecyberice is there any guide on how to build those images ?

breandan, 2018-07-27:
This seems to be a common problem on this release. I was able to compress the necessary images just  enough to fit on the remaining ~87MB of free space on the root parition using `gzip --best` as @sensorii describes above. We just use  `docker pull ${docker_tag} && docker save ${docker_tag} | gzip --best > ${image_filename}` and then copy that to root partition somewhere, however there are [further steps](https://github.com/docker-slim/docker-slim) you could take to reduce the size of the image. There is also ~29MB of free space on the HypriotOS boot partition, so you could plaintext encode the compressed file and write it to the [`write_files`](https://cloudinit.readthedocs.io/en/latest/topics/examples.html#writing-out-arbitrary-files) portion of the cloud-init `user-data` file under on the boot partition. In total you have about ~116MB of free space to play with. The full procedure we use is in [this install script](https://github.com/duckietown/scripts/blob/master/docs/DuckieOS1-RPI3Bp.sh).

breandan, 2018-09-09:
Out of curiosity, is there a way to expand the HypriotOS partition after the image has been flashed and prior to first boot?

firecyberice, 2018-11-10:
@breandan Sure you can use any partitioning tool that understands ext4 to expand the partition and the filessystem after flashing the SD and before putting it into the PI.

closed 2019-09-16 by StefanScherer