hypriot/image-builder-rpi #245: Installing wireguard

vielmetti, 2018-05-17:
I'm installing `wireguard` per the Debian instructions from https://www.wireguard.com/install/ . It's a very nice cross-platform VPN client and server implemented as a kernel module.

After I do

```
sudo apt-get install  wireguard wireguard-dkms wireguard-tools
```

the kernel tries to compile a new module, but fails, as the build scripts are for x86 instead of armhf:

```
Building initial module for 4.4.50-hypriotos-v7+
```

```
./scripts/recordmcount: 1: ./scripts/recordmcount: Syntax error: "(" unexpected
scripts/Makefile.build:258: recipe for target '/var/lib/dkms/wireguard/0.0.20180513-1/build/main.o' failed
```

I've been hacking on this for a bit and tried to rebuild the build scripts, but then ran into other issues. 

Suggestions on how to approach this would be welcome (I don't spend a lot of time in kernel-land).

zx2c4, 2018-05-17:
Looks like the kernel was cross compiled. `cd` to the kernel directory, and then run `make scripts`.

vielmetti, 2018-05-17:
I think this is the same issue fundamentally as https://github.com/hypriot/image-builder-rpi/issues/215

vielmetti, 2018-05-17:
ok, some progress.

```
emv@parviflorus:/usr/src/linux-headers-4.4.50-hypriotos-v7+$ sudo make scripts
[... many things compile ...]
  HOSTCC  scripts/sortextable
scripts/sortextable.c:31:32: fatal error: tools/be_byteshift.h: No such file or directory
 #include <tools/be_byteshift.h>
                                ^
compilation terminated.
scripts/Makefile.host:91: recipe for target 'scripts/sortextable' failed
make[1]: *** [scripts/sortextable] Error 1
```

vielmetti, 2018-05-17:
Turns out I didn't need `sortextable` or anything else that didn't build.

```
wireguard:
Running module version sanity check.
 - Original module
   - No original module exists within this kernel
 - Installation
   - Installing to /lib/modules/4.4.50-hypriotos-v7+/kernel/net/

depmod.....

DKMS: install completed.
Setting up wireguard-tools (0.0.20180513-1) ...
Setting up wireguard (0.0.20180513-1) ...
emv@parviflorus:~$ sudo wg
emv@parviflorus:~$ sudo modprobe wireguard
emv@parviflorus:~$ sudo lsmod | grep wireguard
wireguard             150422  0 
ip6_udp_tunnel          1789  1 wireguard
udp_tunnel              2191  1 wireguard
ipv6                  349938  120 bridge,wireguard
```

Now to get a tunnel up to prove that it Actually Works.

vielmetti, 2018-05-17:
tunnel up! the missing `tools/be_byteshift.h` header did not make a difference in the end.

closed 2018-05-17 by vielmetti

StefanScherer, 2018-05-17:
Great that it's working now. @vielmetti Which steps were needed to fix you issue?

vielmetti, 2018-05-17:
I installed the appropriate kernel headers and (per @zx2c4 suggestion) rebuilt the build scripts, with

```
emv@parviflorus:/usr/src/linux-headers-4.4.50-hypriotos-v7+$ sudo make scripts
```

After that, everything came up according to directions when I did

```
$ sudo apt-get install  wireguard wireguard-dkms wireguard-tools
```

vielmetti, 2018-10-29:
Found this issue again, glad I opened it and solved my own problem. 

The only request is that a next Hypriot release whenever that is should have wireguard shipped by default.

stoffeastrom, 2018-12-09:
@vielmetti Getting the same issue as above. Which are the appropriate kernel headers?

stoffeastrom, 2018-12-10:
Interesting.. I still got the problem where `wireguard-dkms` couldn't be built since the cross compiling. First I tried `sudo make scripts` in 

`/usr/src/linux-headers-4.14.34-hypriotos-v7+`
`sudo apt-get install  --reinstall wireguard wireguard-dkms wireguard-tools`

same problem. Then I also did it in 

`/usr/src/linux-headers-4.14.34-hypriotos+`
`sudo apt-get install  --reinstall wireguard wireguard-dkms wireguard-tools`

and now it worked 🚀. Moving on to see if I can get a tunnel up

vielmetti, 2018-12-12:
Thanks for the update @stoffeastrom - glad you got it working.