hypriot/image-builder-rpi #246: Hypriot 1.9.0 :  E: Package 'acl' has no installation candidate

cgHome, 2018-06-02:
Hello, 

I tried to install the package "acl", unfortunately without success. I get the following error message: 

```
$ sudo apt-get install acl
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package acl is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'acl' has no installation candidate
````

Gruss Chris

StefanScherer, 2018-06-02:
Try `sudo apt-get update` first and repeat the command. Normally the cached package lists are outdated very soon.

cgHome, 2018-06-02:
Okay, this works ! ! THX .... 

But, i use cloud-init and according to my user-data file (see below), the (apt-get) update should be done automatically at the first boot, which seems not to happen. 

### user-data.yml:
```
#cloud-config
#

# The current version of cloud-init in the Hypriot rpi-64 is 0.7.9
# When dealing with cloud-init, it is SUPER important to know the version
# I have wasted many hours creating servers to find out the module I was trying to use wasn't in the cloud-init version I had
# Documentation: http://cloudinit.readthedocs.io/en/0.7.9/index.html

hostname: cflat-srv
manage_etc_hosts: true

resize_rootfs: true
growpart:
    mode: auto
    devices: ["/"]
    ignore_growroot_disabled: false

users: 
  - name: admin
    gecos: "Administrator"
    sudo: ALL=(ALL) NOPASSWD:ALL
    shell: /bin/bash
    groups: users,docker,video
    plain_text_passwd: hypriot
    lock_passwd: false
    ssh_pwauth: true
    chpasswd: { expire: false }

package_update: true
package_upgrade: true
package_reboot_if_required: true

packages:
  - ntp
  - acl

locale: "en_US.UTF-8"
timezone: "Europe/Zurich"

write_files:
    - path: "/etc/docker/daemon.json"
      owner: "root:root"
      content: |
        {
          "labels": [ "os=linux", "arch=arm64" ],
          "experimental": true
        }

# These commands will be ran once on first boot only
runcmd:
  - [ systemctl, restart, avahi-daemon ]
  - [ systemctl, restart, docker ]
  - [ docker, swarm, init ]

  # Set symlink: USB-disk/vol:data -> /var/data
  - sudo ln -sf /dev/disk/by-label/data /var/data
  - sudo setfacl -m u:www-data:rwx /var/data/cloud

  # portainer
  - [
      docker, service, create, 
      "--detach=false", 
      "--name", "portainer", 
      "--publish", "9000:9000", 
      "--mount", "type=volume,src=portainer_data,dst=/data", 
      "--mount", "type=bind,src=//var/run/docker.sock,dst=/var/run/docker.sock", 
      "portainer/portainer", "-H", "unix:///var/run/docker.sock", "--no-auth"
    ]

  # watchtower
  - [
      docker, service, create, 
      "--detach=false", 
      "--name", "watchtower", 
      "--mount", "type=bind,src=//var/run/docker.sock,dst=/var/run/docker.sock", 
      "talmai/rpi-watchtower", "7" 
    ]

  - echo "*** first boot finished ***"
````

StefanScherer, 2019-09-16:
Closing due to inactivity

closed 2019-09-16 by StefanScherer