hypriot/image-builder-rpi #248: KubeletVersion error

obulix, 2018-06-07:
Hi ,

I am try to bring up kubernetes in my RPI 3+..
I have followed the steps specified in the following location.
https://blog.hypriot.com/post/setup-kubernetes-raspberry-pi-cluster/

I am struck with the following error.

# kubeadm init --pod-network-cidr 10.244.0.0/16
[init] Using Kubernetes version: v1.10.4
[init] Using Authorization modes: [Node RBAC]
[preflight] Running pre-flight checks.
        [WARNING SystemVerification]: docker version is greater than the most recently validated version. Docker version: 18.04.0-ce. Max validated version: 17.03
        [WARNING FileExisting-crictl]: crictl not found in system path
Suggestion: go get github.com/kubernetes-incubator/cri-tools/cmd/crictl
[preflight] Some fatal errors occurred:
        [ERROR KubeletVersion]: couldn't get kubelet version: exit status 2
[preflight] If you know what you are doing, you can make a check non-fatal with `--ignore-preflight-errors=...`

I am not sure on what i am missing here.
Any help will be appreciated.

-Dinakaran.AS

jeefberkey, 2018-06-07:
I've been following https://github.com/kubernetes/kubernetes/issues/64234 which I think looked like your issue when I found it, but there is a new release that I haven't tried yet (1.10.4). I have my pis still pinned to 1.10.2.

alexellis, 2018-06-19:
I have an issue with 1.10.2 ..

StefanScherer, 2019-09-16:
Closing due to inactivity

closed 2019-09-16 by StefanScherer