hypriot/image-builder-rpi #249: containerd version without docker

hoshsadiq, 2018-06-08:
Any chance we can get a containerd version without docker-ce and/or docker-machine? Usecase is to have a kubernetes instance that uses containerd without docker. See [this link](https://kubernetes.io/blog/2018/05/24/kubernetes-containerd-integration-goes-ga/).

jeefberkey, 2018-06-08:
Is there a package available?

StefanScherer, 2018-06-08:
Maybe related to #208

hoshsadiq, 2018-06-08:
Yes! Definitely related, however, this is specifically about having a build with just containerd. Arguably, having a Kubernetes ready image makes this redundant, but I think these would be separate packages.  Then again, I do agree with what ulm0 said, which is that hypriot should be as lightweight as possible. IMO if Kubernetes support is provided, users should be provided with cloud-init templates to use. This way users can have a good starting point.

StefanScherer, 2018-06-08:
Well, it's relatively easy to fork our repo and fork the CircleCI pipeline as well with it to try something different, replace some packages and so on. You can do the whole development in the Cloud without local compilation. It's a pattern like I have described in https://stefanscherer.github.io/fork-appveyor-buildpipeline/ - for this rpi repo it's similar, CircleCI even saves the SD card image for each git commit you push to your fork so you can download the image and test it and adjust some fixes, commit again, download and flash again...

I don't know if there's a package for containerd or the official way to install it for Kubernetes. Maybe someone from our community can chime in.

hoshsadiq, 2018-06-08:
Docker runs ontop of containerd, so there has to be some sort of package. I'm still trying to figure out a good cloud-init config for now and still trying to figure out how the rpi works etc. Once I've got a build up and running that I'm happy with, that's when I'll start expirment with improvements. Not quite sure what the consensus was on in #208, so I'm going to ask again; will a PR be accepted if I change it so that we have two different builds and thus two different images (e.g. hypriot `hypriotos-rpi-v1.9.0-docker.img.zip` and `hypriotos-rpi-v1.9.0-containerd.img.zip`)?

StefanScherer, 2019-09-08:
We have no plans to do that.

closed 2019-09-08 by StefanScherer