hypriot/image-builder-rpi #250: NFS Booting

shreddedbacon, 2018-06-16:
I am trying to NFS boot the hypriot image, and it looks like it fails to do cloud-init properly.
I understand this is not a typical usecase, but I'm not sure where to start looking at this.

This is output from serial console. It looks like it tries to get information from `http://169.254.169.254/2009-04-04/meta-data/instance-id` a few times then gives up.
Once it gives up, I get returned to a login prompt after it finishes booting, but the default username/password don't work.

Any ideas?
```
[   18.503942] cloud-init[112]: Cloud-init v. 0.7.9 running 'init-local' at Sat, 28 Apr 2018 18:56:59 +0000. Up 15.92 seconds.                                                                
[  OK  ] Started Initial cloud-init job (pre-networking).
[  OK  ] Reached target Network (Pre).
         Starting Raise network interfaces...                                                                                                                                                 
[  OK  ] Started Raise network interfaces.
         Starting Initial cloud-init job (metadata service crawler)...
[   28.508966] cloud-init[274]: Cloud-init v. 0.7.9 running 'init' at Sat, 28 Apr 2018 18:57:09 +0000. Up 25.40 seconds.                                                                      
[   28.510958] cloud-init[274]: ci-info: +++++++++++++++++++++++++++++Net device info++++++++++++++++++++++++++++++                                                                           
[   28.512903] cloud-init[274]: ci-info: +--------+-------+-----------+---------------+-------+-------------------+                                                                           
[   28.514569] cloud-init[274]: ci-info: | Device |   Up  |  Address  |      Mask     | Scope |     Hw-Address    |                                                                           
[   28.516525] cloud-init[274]: ci-info: +--------+-------+-----------+---------------+-------+-------------------+                                                                           
[   28.518428] cloud-init[274]: ci-info: | eth0:  |  True | 192.168.1.45 | 255.255.255.0 |   .   | b8:27:eb:92:2e:57 |                                                                           
[   28.520700] cloud-init[274]: ci-info: | eth0:  |  True |     .     |       .       |   d   | b8:27:eb:92:2e:57 |                                                                           
[   28.522558] cloud-init[274]: ci-info: | wlan0: | False |     .     |       .       |   .   | b8:27:eb:c7:7b:02 |                                                                           
[   28.524685] cloud-init[274]: ci-info: |  lo:   |  True | 127.0.0.1 |   255.0.0.0   |   .   |         .         |                                                                           
[   28.526643] cloud-init[274]: ci-info: |  lo:   |  True |     .     |       .       |   d   |         .         |                                                                           
[   28.528867] cloud-init[274]: ci-info: +--------+-------+-----------+---------------+-------+-------------------+                                                                           
[   28.530965] cloud-init[274]: ci-info: ++++++++++++++++++++++++++++Route IPv4 info+++++++++++++++++++++++++++++                                                                             
[   28.533440] cloud-init[274]: ci-info: +-------+-------------+------------+---------------+-----------+-------+                                                                             
[   28.535560] cloud-init[274]: ci-info: | Route | Destination |  Gateway   |    Genmask    | Interface | Flags |                                                                             
[   28.537881] cloud-init[274]: ci-info: +-------+-------------+------------+---------------+-----------+-------+                                                                             
[   28.539956] cloud-init[274]: ci-info: |   0   |   0.0.0.0   | 192.168.1.254 |    0.0.0.0    |    eth0   |   UG  |                                                                             
[   28.542501] cloud-init[274]: ci-info: |   1   |   192.168.1.0  |  0.0.0.0   | 255.255.255.0 |    eth0   |   U   |                                                                             
[   28.544692] cloud-init[274]: ci-info: +-------+-------------+------------+---------------+-----------+-------+                                                                             
[   28.547098] cloud-init[274]: 2018-04-28 18:57:12,352 - url_helper.py[WARNING]: Calling 'http://169.254.169.254/2009-04-04/meta-data/instance-id' failed [1/120s]: request error [HTTPConnec
tionPool(host='169.254.169.254', port=80): Max retries exceeded with url: /2009-04-04/meta-data/instance-id (Caused by NewConnectionError('<requests.packages.urllib3.connection.HTTPConnectio
n object at 0x75a5d1f0>: Failed to establish a new connection: [Errno 113] No route to host',))]
... REPEATx? until it gives up and goes to login prompt
```
Could I be missing something from the image? The NFS share is created from the 2nd partition (not /boot) directly from the .img file.
Exactly the same way I mount raspbian images via NFS, with no issue.

shreddedbacon, 2018-06-17:
I figured it out.

If I replace the symlinks for `/var/lib/cloud/seed/nocloud-net/user-data` and `/var/lib/cloud/seed/nocloud-net/meta-data` with the files I want to use, everything works as expected.

closed 2018-06-17 by shreddedbacon

webdavis, 2020-12-29:
> I figured it out.
> 
> If I replace the symlinks for `/var/lib/cloud/seed/nocloud-net/user-data` and `/var/lib/cloud/seed/nocloud-net/meta-data` with the files I want to use, everything works as expected.
> 
> I have to replace both links, even if meta-data is unused, it needs to be an empty file, not a broken symlink

Hi @shreddedbacon, do you have a write up of the steps you took to accomplish this? I'm currently attempting this on my home network and running into a boot issue `Failed to start Raise network interfaces`.

shreddedbacon, 2021-01-03:
Hey @webdavis it has been so long since I did this that I have forgotten where I kept all the information. I'll try and dig through from when I was working on this, as I haven't touched it since.

webdavis, 2021-01-05:
@shreddedbacon, thanks I appreciate it! But no worries if you can't find it, or if you don't have the time.